UtilitechAS/amsreader-firmware #428: Support relative non-rooted URLs

bmork, 2023-02-05:
I'm using a reverse proxy in front of the amsreader web server for external access with TLS etc.  This reverse proxy is shared between a number of similar services.  But for simplicity I'd like them all under the same domain name.  This means that each device behind the reverse proxy ends up with a non-root base URL.

I.e.

https://www.example.com/iot1/ => http://192.168.0.1/
https://www.example.com/iot2/ => http://192.168.0.2/
https://www.example.com/someservice/ => http://127.0.0.1:8833/

etc.  You get the point.

The problem is that the amsreader firmware isn't very happy about this kind of relocation.  It really insisn'ts on being at the root.

I've so far use somewhat of a hack to work aorund this problem (nginx partial config):

```
        location / {

                # PoW-K needs a helping hand...
                if ($http_referer ~ /pow-k/) {
                        rewrite .* /pow-k$uri redirect;
                }
          ...
        }
        location /pow-k/ {
                proxy_pass http://192.168.0.1/;
        }

```
Which has sort of worked with the 2.1.x firmware series, with only this additional helper:
```
        location /save {
                proxy_pass http://192.168.0.1;
        }

```
But the 2.2.x firmware has made this more difficult.  So what I wonder is:  Would it be possible to rewrite the web pages in a way that supported moving the base URL without such hacks?  That would be lovely...

gskjold, 2023-02-16:
Definitely possible, I will look at this for the next minor release

gskjold, 2023-12-26:
Sorry this has taken a while, but I have now added support to define a context path, which would help in this matter.

If you want to test this, here is a build of v2.3 branch. **WARNING**, this is a development build, please make sure you are able to re-flash stable firmware via USB if necessary. Also, back up your config before flashing.
[esp32.zip](https://github.com/UtilitechAS/amsreader-firmware/files/13770272/esp32.zip)
[esp32c3.zip](https://github.com/UtilitechAS/amsreader-firmware/files/13770273/esp32c3.zip)
[esp32s2.zip](https://github.com/UtilitechAS/amsreader-firmware/files/13770274/esp32s2.zip)
[esp32solo.zip](https://github.com/UtilitechAS/amsreader-firmware/files/13770275/esp32solo.zip)
[esp8266.zip](https://github.com/UtilitechAS/amsreader-firmware/files/13770276/esp8266.zip)

closed 2024-01-26 by gskjold