UtilitechAS/amsreader-firmware #432: Several Incorrect Unit of Measurement

nossaile, 2023-02-07:
Several unit of measurements that were correct before are now incorrect! Like:
"var" instead of "VAr", 
"kvarh" instead of "kVArh",
"C" instead of "°C", 

See logfile:

2023-02-07 13:19:35.214 WARNING (MainThread) [homeassistant.components.sensor] Entity sensor.ams_cf58_hourcost (<class 'homeassistant.components.mqtt.sensor.MqttSensor'>) is using state class 'total_increasing' which is impossible considering device class ('monetary') it is using; expected None or one of 'total'; Please update your configuration if your entity is manually configured, otherwise create a bug report at https://github.com/home-assistant/core/issues?q=is%3Aopen+is%3Aissue+label%3A%22integration%3A+mqtt%22
2023-02-07 13:19:35.222 WARNING (MainThread) [homeassistant.components.sensor] Entity sensor.ams_cf58_daycost (<class 'homeassistant.components.mqtt.sensor.MqttSensor'>) is using state class 'total_increasing' which is impossible considering device class ('monetary') it is using; expected None or one of 'total'; Please update your configuration if your entity is manually configured, otherwise create a bug report at https://github.com/home-assistant/core/issues?q=is%3Aopen+is%3Aissue+label%3A%22integration%3A+mqtt%22
2023-02-07 13:19:35.224 WARNING (MainThread) [homeassistant.components.sensor] Entity sensor.ams_cf58_monthcost (<class 'homeassistant.components.mqtt.sensor.MqttSensor'>) is using state class 'total_increasing' which is impossible considering device class ('monetary') it is using; expected None or one of 'total'; Please update your configuration if your entity is manually configured, otherwise create a bug report at https://github.com/home-assistant/core/issues?q=is%3Aopen+is%3Aissue+label%3A%22integration%3A+mqtt%22
2023-02-07 13:19:41.930 WARNING (MainThread) [homeassistant.components.sensor] Entity sensor.ams_cf58_tqo (<class 'homeassistant.components.mqtt.sensor.MqttSensor'>) is using native unit of measurement 'kvarh' which is not a valid unit for the device class ('energy') it is using; expected one of ['Wh', 'MJ', 'GJ', 'MWh', 'kWh']; Please update your configuration if your entity is manually configured, otherwise create a bug report at https://github.com/home-assistant/core/issues?q=is%3Aopen+is%3Aissue+label%3A%22integration%3A+mqtt%22
2023-02-07 13:19:41.937 WARNING (MainThread) [homeassistant.components.sensor] Entity sensor.ams_cf58_tqi (<class 'homeassistant.components.mqtt.sensor.MqttSensor'>) is using native unit of measurement 'kvarh' which is not a valid unit for the device class ('energy') it is using; expected one of ['Wh', 'MJ', 'GJ', 'MWh', 'kWh']; Please update your configuration if your entity is manually configured, otherwise create a bug report at https://github.com/home-assistant/core/issues?q=is%3Aopen+is%3Aissue+label%3A%22integration%3A+mqtt%22
2023-02-07 13:20:11.463 WARNING (MainThread) [homeassistant.components.sensor] Entity sensor.ams_cf58_temp (<class 'homeassistant.components.mqtt.sensor.MqttSensor'>) is using native unit of measurement 'C' which is not a valid unit for the device class ('temperature') it is using; expected one of ['°F', '°C', 'K']; Please update your configuration if your entity is manually configured, otherwise create a bug report at https://github.com/home-assistant/core/issues?q=is%3Aopen+is%3Aissue+label%3A%22integration%3A+mqtt%22
2023-02-07 13:20:12.883 WARNING (Recorder) [homeassistant.components.sensor.recorder] The unit of sensor.ams_cf58_q (var) can not be converted to the unit of previously compiled statistics (VAr). Generation of long term statistics will be suppressed unless the unit changes back to VAr or a compatible unit. Go to https://my.home-assistant.io/redirect/developer_statistics to fix this
2023-02-07 13:20:12.886 WARNING (Recorder) [homeassistant.components.sensor.recorder] The unit of sensor.ams_cf58_qo (var) can not be converted to the unit of previously compiled statistics (VAr). Generation of long term statistics will be suppressed unless the unit changes back to VAr or a compatible unit. Go to https://my.home-assistant.io/redirect/developer_statistics to fix this
2023-02-07 13:20:12.888 WARNING (Recorder) [homeassistant.components.sensor.recorder] The unit of sensor.ams_cf58_tqi (kvarh) can not be converted to the unit of previously compiled statistics (kVArh). Generation of long term statistics will be suppressed unless the unit changes back to kVArh or a compatible unit. Go to https://my.home-assistant.io/redirect/developer_statistics to fix this
2023-02-07 13:20:12.889 WARNING (Recorder) [homeassistant.components.sensor.recorder] The unit of sensor.ams_cf58_tqo (kvarh) can not be converted to the unit of previously compiled statistics (kVArh). Generation of long term statistics will be suppressed unless the unit changes back to kVArh or a compatible unit. Go to https://my.home-assistant.io/redirect/developer_statistics to fix this
2023-02-07 13:20:12.890 WARNING (Recorder) [homeassistant.components.sensor.recorder] The unit of sensor.ams_cf58_pf (%) can not be converted to the unit of previously compiled statistics (). Generation of long term statistics will be suppressed unless the unit changes back to  or a compatible unit. Go to https://my.home-assistant.io/redirect/developer_statistics to fix this
2023-02-07 13:20:12.890 WARNING (Recorder) [homeassistant.components.sensor.recorder] The unit of sensor.ams_cf58_pf1 (%) can not be converted to the unit of previously compiled statistics (). Generation of long term statistics will be suppressed unless the unit changes back to  or a compatible unit. Go to https://my.home-assistant.io/redirect/developer_statistics to fix this
2023-02-07 13:20:12.891 WARNING (Recorder) [homeassistant.components.sensor.recorder] The unit of sensor.ams_cf58_pf2 (%) can not be converted to the unit of previously compiled statistics (). Generation of long term statistics will be suppressed unless the unit changes back to  or a compatible unit. Go to https://my.home-assistant.io/redirect/developer_statistics to fix this
2023-02-07 13:20:12.891 WARNING (Recorder) [homeassistant.components.sensor.recorder] The unit of sensor.ams_cf58_pf3 (%) can not be converted to the unit of previously compiled statistics (). Generation of long term statistics will be suppressed unless the unit changes back to  or a compatible unit. Go to https://my.home-assistant.io/redirect/developer_statistics to fix this

As you can see there are more than UoM to correct.

Power Factor could be both with ("%") and without UoM. Both are correct, but why change? And my suggestion: Remove them, because as I understand it they are not part of the payload from Pow-P1 and they can be calculated in HA if needed.

**Hardware information:**
 - Country: Sweden
 - Meter: L&G 320L
 - Encryption enabled No
 - AMS reader: Pow-P1

**Relevant firmware information:**
 - Version: v2.2.5
 - MQTT: Yes (Mosquito)
 - MQTT payload type: Home Assistant
 - HAN GPIO: ?
 - HAN baud and parity: 115200 8N1
 - Temperature sensors None
 - ENTSO-E API enabled: [Yes

ArnieO, 2023-02-07:
Yeah, doesn't look pretty - but there was a reason for the change.
I can't find the thread right now but I think I remember someone reported some kind of compatibility issues in Home Assistant.

dagbdagb, 2023-02-07:
I had a good chunk of warnings which simply went away after updating HA OS and front-end.

gskjold, 2023-02-08:
This was changed to comply with https://developers.home-assistant.io/docs/core/entity/sensor/#available-device-classes

See #397 and #408 

Power factor is not available for you because your meter is not reporting it.
Month max will not be the same for Norwegian users who has a 3-peak average.

closed 2023-02-08 by gskjold