UtilitechAS/amsreader-firmware #433: Adding Kaifa MA309M (Austria) as know Hardware configuration

coffinbearer, 2023-02-08:
I recently got my Pow-U+ and it works as intended with my Kaifa MA309M H4LAT1 meter.
Therefore i would like it if the meter could be added to the know hardware configurations
an be recognized from the ams reader firmware as well
[image]

This are the settings for the ams reader meter configuration:

Manufacturer | Baud | Parity | Grid operator | Encryption
-- | -- | -- | -- | --
Kaifa MA309M | 2400 | 8N1 | Netz Niederösterreich GmbH | Yes (Encryption key only)

What information do you need from me (my meter) to be able recognize it?

ArnieO, 2023-02-09:
Hi and thank you for reporting this! 
Could you do a Telnet debug [as described here](https://amsleser.no/blog/post/24-telnet-debug), and copy/post the output in text format? This will allow us to analyze the raw data and see how the meter presents itself.

coffinbearer, 2023-02-09:
Hi, here the verbose debug output:
[amsleser_kaifa_MA309M_austria.txt](https://github.com/UtilitechAS/amsreader-firmware/files/10699997/amsleser_kaifa_MA309M_austria.txt)

ArnieO, 2023-02-10:
Thank you - perfect!

gskjold, 2023-03-27:
Added detection for manufacturer for this meter. Meter ID is also present, but is not annotated with OBIS code. I've seen other meters like this, so I have added code to extract it.

Razzer09, 2023-09-22:
If there is only an Encryption key needed do I leave the Authentication key at "00000000000000000000000000000000"?

Wupsman, 2023-09-23:
Hi Razzer09, yes leave the zeros in, it worked that way for me at EVN too.

Razzer09, 2023-09-23:
Thank you for the quick response. 

Unfortunately this means I don't know why it is not working for me.  :-(

My Setup is:

[image]

) From EVN NetzNÖ

) ESP8266MOD with the following settings:
[image]

) MBUS Adapter: 
https://www.amazon.de/MDYBF-TTL-Zu-MBUS-Modul-Schnittstelle-MBUS-Slave-Modul-Signalisolierung-gr%C3%BCn/dp/B0BZY6T7B7/ref=sr_1_1?keywords=instead+of+mbus&qid=1695458721&sr=8-1

----
I downloaded the Firmware for the ESP8266MOD from here:
[image]
I flashed it using esptool.py through windows command line.

I used the middle two cables from my RJ12 cable:
[image]

Connected the RJ12 cables to the MBUS Board:
[image]

Connected the MBUS Board to the ESP8266MOD:
[image]

[image]

---

My guess is that there is something wrong with this MBUS Adapter Board.

Do I have to change anything in the code or should the precompiled file I downloaded be sufficient?

Wupsman, 2023-09-23:
I would try 8E1 and disable the HAN pullup, reboot in between. 
It is also recommended by amsreader to use the hardware serial, so I would also try with this, will probably be UART0 / WEMOS PIN RX.
https://github.com/UtilitechAS/amsreader-firmware/wiki/GPIO-configuration#han

I also had the current release version v2.2.21 running on an ESP32 Olimex and it only worked with 8E1, now I have it running with the M5 PoESP32 + Farley TTL to MBUS --> here it works with 8E1 and 8N1.
https://github.com/UtilitechAS/amsreader-firmware/issues/365#issuecomment-1712878848
https://github.com/UtilitechAS/amsreader-firmware/files/12452534/esp32.zip

Razzer09, 2023-09-23:
Thank you for all your quick and helpful feedback. I tried:

) Setting 8E1 & disableling HAN pullup -> no luck
) Changing the pins VIN from 3V to 5V, TX Adapter to RX on WEMOS ESP8266 and setting UART0 in Software.
[image]
[image]

--> no luck 

Ich checked on Amazon and it says that the adapter should accept 3.3V and 5V.
[image]

) Tried everything again with an ESP32solo Nodemcu V1.3 on UART0 if I am not mistaken? --> no luck
[image]
[image]

) Finally I did for both ESPs a Telnet debug where they look both pretty much the same. 
[image]

Since there is no data shown in the debug I assume it must be the adapter is either broken or does not work with ESP.

I would try this board instead or can you recommend anything that works well with ESP8266?
https://www.amazon.de/Clickboard-Daughter-Boards-Module-MIKROE-4137/dp/B0BRKTGBBT/ref=sr_1_1?__mk_de_DE=%C3%85M%C3%85%C5%BD%C3%95%C3%91&crid=297UXP0E13Y70&keywords=m-bus-slave-click&qid=1695486105&sprefix=m-bus-slave-click%2Caps%2C96&sr=8-1

Wupsman, 2023-09-23:
It's a pity that no data arrives here. I have two of the Farley TTL to MBUS running without any problems so far, the ELKO of one was damaged by the delivery and after the exchange it also worked, but I have Sagemcom T210-D Smartmeter. 
Did the interface from your Samartmeter work before?

dbeinder, 2023-09-24:
@Razzer09 
Did you measure the MBUS voltage with a multimeter? It should be ~27V with regular drops to ~15V during data transmission. Since the data rate is so slow, you can easily check if the interface is active with a cheap multimeter. Just to be sure, you did request the activation of the MBUS port at your grid company?

Razzer09, 2023-09-25:
@dbeinder 
Thank you for the idea. I checked my Smartmeter and got exactly 27V with regular 5 second drops to around 19V. This also verified that I was using the correct cables of the RJ12 cable because on the other 2 cables I got 0V.

I requested the encryption key from my grid company and assumed that the activation is included with it. But seeing how the voltage oscillate means it is already activated right?

dbeinder, 2023-09-25:
@Razzer09 
Looking at the opto-coupler datasheet, I think you are using the wrong pin on the MBUS board, try the one labeled RX:
[image]
In case it doesn't work and you're considering the mikroe board, I could also send you [one of my boards](https://github.com/UtilitechAS/amsreader-firmware/discussions/480) ;)

Razzer09, 2023-09-25:
OMG it worked!

I tried connecting:
) RX from MBUS board to RX from ESP8266 and UART0 set in Software (no pullup) --> Didn't work
) RX from MBUS board to TX from ESP8266 and UART0 set in Software (no pullup) --> Didn't work
) RX from MBUS board to RX from ESP8266 and UART2 set in Software (no pullup) --> Didn't work
) RX from MBUS board to TX from ESP8266 and UART2 set in Software (no pullup) --> Didn't work

Desperately I finally tried:
) RX from MBUS board to D1 from ESP8266 and GPIO5 set in Software (no pullup) --> **IT WORKED** :-D
[image]

@dbeinder Thank you so much. 👍

ArnieO, 2023-09-26:
It is great to see the community stepping in like this! 😀