UtilitechAS/amsreader-firmware #434: To high supply voltage after upgrade to 2.2.6 is shown

Lin4fun, 2023-02-10:
**Describe the bug**
After upgrade to FW 2.2.6 a to high supply voltage shown 4.24 volt. See attached picture.

**Screenshots**
[image]
**Hardware information:**
 - Country: Norway
 - Meter: Kamstrup
 - Encryption enabled [no]
 - AMS reader: Pow-U+, ESP32S2
 - M-bus adapter (if applicable):

**Relevant firmware information:**
 - Version: 2.2.6
 - ENTSO-E API enabled: yes

gskjold, 2023-02-10:
Weird... could you download a config file and attach here? You can untick everything except gpio

warfair1337, 2023-02-10:
Having the same issue here after upgrade to 2.2.6, 4.10V.

gskjold, 2023-02-10:
Also, what version did you upgrade from?

warfair1337, 2023-02-10:
Upgraded from 2.2.0 to 2.2.5, and then 2.2.6 45 min later. 

`amsconfig
version v2.2.6
boardType 7
gpioHanPin 16
gpioApPin 0
gpioLedPinRed 13
gpioLedPinGreen 14
gpioLedRgbInverted 1
gpioVccPin 10
gpioVccOffset 0.00
gpioVccMultiplier 1.000
gpioVccBootLimit 0.0
gpioVccResistorGnd 22
gpioVccResistorVcc 33
dayplot 5 1676048411 66218820 0 1257 1029 702 1380 470 790 535 1483 1640 1279 1950 1541 2802 1410 2287 1262 3531 1470 1049 1440 1149 1211 2144 858
monthplot 6 1675983660 66192614 1 25500 35280 33690 36080 46970 88730 34700 33680 28410 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 69910
energyaccounting 5 2 27.30 479 198 0.00 0 0 10 3.53 8 3.01 9 2.33 6 14.09 5 4.62 `

gskjold, 2023-02-10:
config looks sane. How about /data.json ?

warfair1337, 2023-02-10:
Output from /data.json

`{"im":22308,"om":0,"mf":56,"i":1990,"e":0,"ri":0,"re":747,"ic":0.377,"ec":0.000,"ric":0.000,"rec":0.209,"u1":235.50,"u2":235.60,"u3":235.50,"i1":5.70,"i2":5.65,"i3":3.20,"f":0.00,"f1":0.00,"f2":0.00,"f3":0.00,"v":4.101,"r":-72,"t":-127.00,"u":1057,"m":112280,"em":3,"hm":1,"wm":1,"mm":1,"me":0,"p":0.88,"mt":1,"ds":1,"ea":{"x":7.4,"p":[3.53,14.09,4.62],"t":10,"h":{"u":0.38,"c":0.33,"p":0.00,"i":0.00},"d":{"u":26.58,"c":26.29,"p":0.00,"i":0.00},"m":{"u":389.62,"c":505.29,"p":0.00,"i":0.00}},"pr":"10YNO-1--------2","he":0,"ee":0,"c":1676051433,"a":true}`

warfair1337, 2023-02-10:
In my setup the POW-U is always connected with USB-power.
A friend of mine who received his POW-U today (came with 2.2.5) just upgraded to 2.2.6 and is having the same issue. Also on Aidon-meter and USB-powered.

gskjold, 2023-02-10:
Fixed in 2.2.7

warfair1337, 2023-02-10:
All good in 2.2.7. Thanks 👍🏻

gskjold, 2023-02-10:
Excellent!

closed 2023-02-10 by gskjold