UtilitechAS/amsreader-firmware #435: Some trouble updating from 2.2.5 to 2.2.7

tronde-ams, 2023-02-10:
ESP32 configured as Lolin D32. MQTT. aidon 3p / IT Norway, own Entso-e api key.

I have to units with same config, except different MQTT broker.

One was troublefree, the other not.

I clicked om the "New version v2.2.7" and it started the update process as expected. 
It did not load 2.2.7, but started again as 2.2.5.
I tried several times with two different PCs and browswers.

Then I checked the device information and found this.
I tried to update to 2.2.6, and it went OK
[image]

Then I tried to update to 2.2.7, and it was OK too, except I got this
[image]

After a reboot, it disappeared, 

Prices did not update before new hour, but i don't pay much attention to that, because I know prices can be kind of shaky.

Main problem was the lack of update to new version.
No big deal for me, but I report it since it was not expected to happen.

gskjold, 2023-02-11:
Interesting. First glance in the logs suggest that any 2.2.5 upgrade was offered 2.2.7. As far as I can tell there was never any repeat upgrade requests from same device, but we'll see if anyone else have experienced this

tronde-ams, 2023-02-11:
My unit was also offered 2.2.7 on the main page, but it did not install. Then I looked in the information menu and found the 2.2.6 as I show above.

laageide, 2023-02-12:
I had the same issiue. The solution for me were to reboot the AMS-reader then install update.

gskjold, 2023-03-11:
My assumption is that this is some network related issue. We rely on esp-idf framework for communication, so if the issue is internal, it is outside our code.

tronde-ams, 2023-03-27:
Can you see something in the logs for around 18:40 today? Same trouble with one unit trying to upgrade from .10 to .11. I tried twice. Had to download the file and upload it. I don't rememer if it was the same unit.

gskjold, 2023-03-28:
18:40 there was two firmware request from ESP32 with mac address ending with F1:50, where 2.2.11 was offered to the device.

Does it reboot at all when this happens? Starting from 2.2.11, you can find the reason for last reboot, which could be useful next time this happen:
[image]

ArnieO, 2023-03-28:
I see that I sometimes need to try the one-click-upgrade more than once. Judging from customer feedback I get, this is not uncommon. I advise (and describe how to do) a "manual" update (download bin file etc) if repeated attempts fail, I had such a case this morning (a customer in Austria). 

Would it be possible for the firmware to detect that the one-click-upgrade failed, and if so advise user to retry - or do a manual upgrade?

tronde-ams, 2023-03-28:
F1:50 is my unit. 
Nothing unusual as long as I can see, except it did not load 2.2.11.

It is not a problem for me, but I report when I experience unexpected behavior.

ArnieO, 2023-03-28:
> It is not a problem for me, but I report when I experience unexpected behavior.

Thank you for reporting! There seems to be some snags related to upgrading, not sure what the reason could be. Timeout during the process, maybe? (But just guessing now.)

I have today advised a Swedish user that wanted to upgrade a Pow-K+ from 2.1.14. He bought his amsleser in November-22 and has never managed to upgrade; both one-click and "manual" (download binary etc) has failed. 

I attempted to reproduce the problem and (again) had to try one-click upgrade several times. The strange thing was that it first upgraded to 2.1.17 before accepting upgrade to 2.2.11. (Identical result two times, so maybe by design, @gskjold ?)

Binary download and "manual" upgrade from v2.1.14 also needed several attempts, both me and customer saw this error message during the attempts:
[image]

As mentioned above, it would be great if the firmware would be able to handle unsuccessful upgrades a bit better - if possible give user some guidance on what to do next.

If both one-click and binary upload fails, all newer amsleser devices with ESP32-S2 micro controller can be reprogrammed using the USB cable. The procedure is explained in chapter 9 of Firmware User Manual, that can be downloaded from the product page: https://amsleser.no/index.php?controller=attachment&id_attachment=45

tronde-ams, 2023-03-30:
I see this now for the unit that did not upgrade. 

Last boot: Software reset CPU (12)

(About 1h ago).
My other unit fed from the same power source has not re-booted.

Both have been stable, so re-boot is unexpected.
Any connection, or just a random mishap?

gskjold, 2023-04-15:
Increase timeout from default 8s to 60s. Also added debugging on status page if the expected firmware version doesn't match.

closed 2023-04-15 by gskjold