UtilitechAS/amsreader-firmware #437: Read meter registers on startup

rogerthn2019, 2023-02-11:
**Is your feature request related to a problem? Please describe.**
I'm using `curl --silent http://192.168.C.D/data.json  | awk -F "," '{print $4,$5,$8,$9 ...`
to update some Domoticz devices, including Accumulated import and export and those values are 0 after a restart

**Describe the solution you'd like**
Read meter registers at startup unless there is a trick to read those before next hour change?

**Describe alternatives you've considered**
Force reading of meter registers

**Additional context**
[image]

gskjold, 2023-02-11:
This is a one way communication where meter push data on a schedule. Pulling data is not an option

closed 2023-02-11 by gskjold

rogerthn2019, 2023-02-11:
OK, make sense.