UtilitechAS/amsreader-firmware #439: Disable pullup on HAN pin

dbeinder, 2023-02-11:
**Is your feature request related to a problem? Please describe.**
I'd like to use level shifter with relatively high output impedance, and the Serial library automatically sets the ESP pullup (~45k).

**Describe the solution you'd like**
Either simply disable the pullup or have an option to configure it. Are there any known hardware configurations that depend on the pullup? e.g. the often used TSS721A shouldn't need it. I can implement this myself, which would you prefer?

ArnieO, 2023-02-12:
From memory the internal pull resistors are somewhere in the range 30 to 45 kohm - and as far as I know not very precise. 

As this project is quite mature, and used on many different hardware platforms, we cannot risk to simply disable it. And as you are the first of more than 1000 users to make this request: If you have the possibility to make the code modification yourself to suit your specific needs, I think that is probably the best solution.

dbeinder, 2023-02-22:
I'll probably have a couple of boards made for friends and family, so it would be nice to have this option integrated. Would you be open to a pull request for a custom board type?

closed 2023-02-22 by dbeinder