UtilitechAS/amsreader-firmware #448: Wrong display of import value with multiplicator

Ainnerst, 2023-02-14:
**Describe the bug**
When watt is multiplied by 100 it displays a random value for import but sometimes display correct value. If multiplicator is set to 1 it shows ''correct'' value

**To Reproduce**
Set multiplicators for watt  and A to 100

**Expected behavior**
Show correct value after multiplication

**Screenshots**
[image]
[image]
[image]

**Hardware information:**
 - Country: Norway
 - Meter: Kaifa MA304T4
 - Encryption enabled no
 - AMS reader: Pow-U, esp32s2

**Relevant firmware information:**
 - Version: 2.2.8
 - MQTT: no
  - HAN baud and parity: 2400 8E1
 - Temperature sensors no
 - ENTSO-E API enabled: yes

ArnieO, 2023-02-14:
Are you then comparing with the value on the LCD screen of your Kaifa meter?
In an installation with many thermostats, the power value will fluctuate quite a bit between subsequent readings.

Ainnerst, 2023-02-14:
> Are you then comparing with the value on the LCD screen of your Kaifa meter? In an installation with many thermostats, the power value will fluctuate quite a bit between subsequent readings.

Yes, and if i set multiplicator back to 1 the value is stable.

ArnieO, 2023-02-14:
> Yes, and if i set multiplicator back to 1 the value is stable.

OK, interesting. I tried to change my multiplier to 100 but did not see anything strange. 
Could it be an overflow error here, @gskjold ?

gskjold, 2023-02-14:
Strange indeed. I will have to look into this, but I cannot say I have seen this before.

May I ask what kind of installation you have that require a 100 multiplier? Is it a meter with current transformer? Kaifa with NVE format should be supported out of the box. But I see you have set a 200A main fuse, so I assume this is not a normal house? :)

gskjold, 2023-02-14:
> > Yes, and if i set multiplicator back to 1 the value is stable.
> 
> OK, interesting. I tried to change my multiplier to 100 but did not see anything strange. Could it be an overflow error here, @gskjold ?

Indeed, exactly what it is! Obviously need to do something about that.

Ainnerst, 2023-02-14:
> 

if I change both watt and amp, it happends more frequently than only changing watt.

Ainnerst, 2023-02-14:
> Strange indeed. I will have to look into this, but I cannot say I have seen this before.
>
> May I ask what kind of installation you have that require a 100 multiplier? Is it a meter with current [...]

Yes, this is a meter with a current transformer, and it is an installation at my workplace where we plan to implement some smart regulation programming to better control different  loads. The 100 multiplier is necessary to account for the current being measured by the transformer.

Ainnerst, 2023-02-14:
I forgot to add that the history for energy use last 24h looks like this ( i assume its the same issue)
[image]

gskjold, 2023-02-14:
Are you comfortable flashing and running a beta?

Make sure you have necessary multiplier on both watt and kwh to get the correct values in the bar chart.

[esp32s2.zip](https://github.com/UtilitechAS/amsreader-firmware/files/10736117/esp32s2.zip)

gskjold, 2023-02-14:
I see now that you have 100 on all the multipliers, which sort of makes sense, but check that the kwh display in AMS matches what is displayed on the meter. I know that in some cases the meter adjusts for kwh reading and no multiplier is necessary

Ainnerst, 2023-02-14:
> I see now that you have 100 on all the multipliers, which sort of makes sense, but check that the kwh display in AMS matches what is displayed on the meter. I know that in some cases the meter adjusts for kwh reading and no multiplier is necessary

Yes, I havent checked that since the numbers are all over the place atm.

Ainnerst, 2023-02-14:
> Are you comfortable flashing and running a beta?
> 
> Make sure you have necessary multiplier on both watt and kwh to get the correct values in the bar chart.
> 
> [esp32s2.zip](https://github.com/UtilitechAS/amsreader-firmware/files/10736117/esp32s2.zip)

not so comfortable with flashing, but can have a look at it when I'm back at work tomorrow

gskjold, 2023-02-15:
Connect USB power to Pow-U, unzip the attached file, go to status page and click "Select firmware file for upgrade", select "firmware.bin" and click upload. After a little while it should reboot with version "1f5a04e". Note that beta firmware will always claim that latest stable is a newer firmware.

[image]
[image]
[image]
[image]
[image]

Ainnerst, 2023-02-15:
> 

That looks alot better :) Both Import value and energy use last 24 hours are looking good! Thanks!

gskjold, 2023-02-16:
Awesome! Will be included in a future patch, but I have yet to decide the release date for that patch. In any case, you just run this beta until 2.2.9 is released

spinehead, 2023-02-21:
Hi, I experience the same sympthoms in my installation (also workplace use, 400A main breaker) 
Can I use the same beta firmware for my reader?

Hardware information:

Country: Norway
Meter: Aidon 6540
Encryption enabled no
AMS reader: Pow-U, esp8266 (GPIO12)

Relevant firmware information:
Version: 2.2.8
MQTT: no
HAN baud and parity: 2400 8E1
Temperature sensors no
ENTSO-E API enabled: yes

gskjold, 2023-02-21:
Attached is a ESP8266 version of the latest snapshot which includes this fix. Unzip and install the same way as above.

[esp8266.zip](https://github.com/UtilitechAS/amsreader-firmware/files/10790511/esp8266.zip)

closed 2023-03-11 by gskjold