UtilitechAS/amsreader-firmware #451: PowK+ continuously crashing

tb205gti, 2023-02-15:
The PowK+ keeps on crashing and restarting, It only runs for 2min before rebooting/restarting. (uptime never gets above 2 min, very often it is just "a few seconds". There have been no changes to the config nor the meter (That I know of). It started today at ~10AM. Running sw 2.1.7. ESPvoltage is 3.30V, free mem ~120Kb. It uses MQTT, NTP and prices, connected to wifi with DHCP assigned address.

I'm reluctant to start a firmware upgrade as I fear it will reboot during the update and leave me with a brick.

**Hardware information:**
 - Country: Denmark
 - Meter: Kamstrup
 - Encryption enabled yes
 - AMS reader: Pow-K+
 - M-bus adapter (if applicable):

**Relevant firmware information:**
 - Version: 2.1.7
 - MQTT: yes
 - MQTT JSON
 - HAN GPIO: GPIO16
 - HAN baud and parity: 2400 8N1
 - Temperature sensors 
 - ENTSO-E API enabled: yes

ArnieO, 2023-02-15:
Hi @tb205gti: This sounds like an "After sale service" case for the product Pow-K+, and not a firmware issue. 
Please contact me on post@amsleser.no to handle it. If the unit is failing, we will send you a new one.

tb205gti, 2023-02-15:
It might already be solved. One of my AP's (Apple Airport Extreme) had the fan running full power for some reason. I rebooted the AP and the PowK+ jumped to one of the other AP's - now it has been running perfect for 14 minuttes.. Strange, but it looks like there is a connection

closed 2023-02-15 by tb205gti

ArnieO, 2023-02-16:
Maybe the "Reboot on problems" option was activated - and caused this?
[image]

tb205gti, 2023-02-17:
I do  not have that option, but it kills like that could be the reason

tb205gti, 2023-02-17:
But it is still acting up - it has lost all price information. I do have the right API key in the config, and I do not see any issues. Is there a way to test then entsoe api ?

ArnieO, 2023-02-18:
Delete the API key from Entso-E. 
We have set up an intermediate server, if the API key is empty it will fetch price data from there in stead.