UtilitechAS/amsreader-firmware #453: Someone get it work in switzerland with ckw?

bqstony, 2023-02-16:
Has someone it succesfull running with the encription key. 

I get actually the error: (W) (errorBlink)(C0) No HAN data received last 30s, single blink

**Hardware information:**
 - Country: switzerland
 - Meter: Kamstrup 8660
 - Encryption yes
 - AMS reader: Pow-k 

**Relevant firmware information:**
 - Version: 2.2.28
 - MQTT: yes
 - HAN baud and parity:  2400 8N1

bqstony, 2023-02-16:
Should i get data without encrption key entered?

I am also wondering why my key is 35 char long and not 33 chars long

ArnieO, 2023-02-16:
This sounds like your meter is not configured correctly. 
You need to contact your grid operator to have it configured for Data push, and obtain the encryption keys that consists of two keys, as explained here: https://amsleser.no/blog/post/22-danmark-encryption-keys

In most places the configuration of your meter can be done remotely, but apparently not all grid companies have the Kamstrup software tool to do that. One Pow-K customer in Switzerland was in this situation (wrong config, wrong key). The grid company obtained the right keys from Kamstrup, and they sent a technician to his house to modify the config of his meter by connecting via the optical port (using a PC etc).

bqstony, 2023-02-17:
@ArnieO thank you for your respond. I will contact the grid company again. Could be they give me the wrong key.

Do you think the HAN-Interface is not activated when it showes the error `No HAN data received last 30s`?

ArnieO, 2023-02-18:
It is not only the wrong key, the meter has to be configured for Data push. 
"No HAN data received last 302" does not necessarily mean the HAN port is activated.

protobock, 2023-02-20:
> In most places the configuration of your meter can be done remotely, but apparently not all grid companies have the Kamstrup software tool to do that. One Pow-K customer in Switzerland was in this [...]

I am that Swiss customer. I can confirm the meter was not in PUSH mode. The blue LED did not blink at all (it should blink blue every 10 seconds). Only a red LED was blinking. The guy at the grid company was not aware of the various modes (PUSH and PULL). By default the meter was in PULL mode. He just thought I needed the GPK11 encryption key they provide when you use Kamstrup's own module (which works in PULL mode).
The guy was kind enough to directly contact Kamstrup to have procedure how to modify my config and to get the GPK60 and GPK61 pair of keys I needed.
And as ArniO said, a technician had to physically come to my home to change my config. As soon as he clicked on "DLMS PUSH 1" button in his software the blue LED started to blink as it should. I had not entered the GPK keys yet in AMS reader's UI so the data were still encrypted but the blue LED confirmed that the signal was correctly pushed to the Pow-K from the meter.
My Pow-K has now been working rock solid for one month.

ArnieO, 2023-02-20:
> I had not entered the GPK keys yet in AMS reader's UI so the data were still encrypted but the blue LED confirmed that the signal was correctly pushed to the Pow-K from the meter. My Pow-K has now been working rock solid for one month.

Our implementation of the blue LED on Pow-K, Pow-U and Pow-P1 is to give a visual indication of the datagram coming from the meter - which proves to be very useful in these cases. 

Good to hear that your Pow-K is running as it should, @protobock !

closed 2023-03-11 by gskjold

bqstony, 2023-03-31:
I would like to write a summary for those who may have the same problem.

First of all, it is important to find the right person at the electricity provider.

The grid operator CKW can make the change remotely in 5 minutes.

1. the grid operator has to configure the smart meter to push mode. He has to send  `DLMS PUSH` (Then HAN turns green in the UI)
2. Generate keys GPK60 (encryption) and GPK61 (authentication) (Enter the Keys in AMS reader UI)

Thats it 😀

Now it works also on my meeter. 
Thx to @protobock & @ArnieO

diaznet, 2023-10-30:
@bqstony thanks for writing this here about CKW. Did you ask to configure PUSH mode via the regular contact address (zaehlen.messen@ckw.ch) ?

bqstony, 2023-11-03:
@diaznet Hi
I had contact with Andre Rast -> https://www.linkedin.com/in/andr%C3%A9-rast-a5aaa31a5 It seems that not everyone knows how to do it. So he is the one who knows. 

I dont want to put his phone number here, but you can find the number in this official paper :)

https://www.sem.admin.ch/dam/metas/de/data/GesetzlichesMesswesen/messen-regeln-sicherstellen/Svs-register%20(13).pdf.download.pdf/Svs-register%20(13).pdf

haldi4803, 2024-01-08:
This kind of thingy?
[image]

ArnieO, 2024-01-09:
> This kind of thingy?

This is a Kamstup Omnipower meter. Our [Pow-K](https://www.amsleser.no/hjem/1-11-pow-k-kamstrup.html#/11-pow_k_options-pow_k_uten_3d_printet_holder_trenger_ikke_eller_printer_selv) is designed especially for this meter. It is installed by removing the lid at the top of your photo. You need to request your grid company to activate the HAN port and configure it for "Data push".

haldi4803, 2024-01-11:
> but we have one customer in Switzerland where the grid company had to do the configuration on-site:

CKW?
Cuz they old me via Mail that the device is "too old" and needs to be replaced^^

ArnieO, 2024-01-11:
> CKW?

I don't know the grid company, he is situated in the Vaud region

The Omnipower 684-11 are definitively not "too old"; this is the model I have myself here. 
So if they cannot activate the port there must be some other reason for that.

protobock, 2024-01-11:
> > CKW?
> 
> I don't know the grid company, he is situated in the Vaud region
> 
> The Omnipower 684-11 are definitively not "too old"; this is the model I have myself here. So if they cannot activate the port there must be some other reason for that.

I am this guy.
My company is a local provider called VOEnergies.
They do not use the usual software (I don't remember the name of this software) to manage the meters because the licence is too expensive. That is the reason why they have limited remote capabilities and they had to come physically to my home.
The technician connected his laptop to my meter with the optical interface and the wholw operation took a couple of minutes.
See my post here:
https://github.com/UtilitechAS/amsreader-firmware/issues/453#issuecomment-1437523551

ArnieO, 2024-01-11:
> They do not use the usual software (I don't remember the name of this software) to manage the meters because the licence is too expensive.

OK - that makes sense.
The name of that software is Omnisoft VisionAir, there is a screenclip from it on page 17 of this document: https://documentation.kamstrup.com/docs/HAN_NVE/en-GB/Information_sheet/CONTAAA62C8B611D4C4EB3186FD81AEE0152/

Kamstrup Omnipower is one of 3 brands that were rolled out here in Norway. In several cases, small grid companies here created regional collaborations (organised as separate entities / joint ventures) that initially handled the rollout/installation of the smart meters - and later takes care of the management of them, including activation and configuration of the HAN port. That organisation model seems to have been quite successful, as the introduction of smart meters require competence that small grid companies do not have, and cannot afford to hire on their own. 

We can log in on "My page" on the gird company website and order activation of the HAN port online, so the process has become quite efficient.

haldi4803, 2024-01-11:
Matter of fact.
I got Hardware version 1 with firmware X1 which does not support the HAN module.
Therefore it will be replaced :)
Will report afterwards.

haldi4803, 2024-01-22:
CKW send a worker who replaced the old device with a new one.... looks literally the same except for the bottom part when you remove the cover of the connectors.
He say CKW will probably replace all devices (was built 2013) "soon".

But oh well, i got it connected to WiFi, did a Firmware update. Seems to work pretty well.

Except the prices^^ entso claims 8cents per kwh.
[image]
More like 17.4 cents for energy and 8.7cents for transport. + additional 2.3 cents for green energy.

ArnieO, 2024-01-22:
> Except the prices^^ entso claims 8cents per kwh.

Are the [prices published on Entso-E website ](https://transparency.entsoe.eu/transmission-domain/r2/dayAheadPrices/show?name=&defaultValue=false&viewType=GRAPH&areaType=BZN&atch=false&dateTime.dateTime=22.01.2024+00:00|CET|DAY&biddingZone.values=CTY|10YCH-SWISSGRIDZ!BZN|10YCH-SWISSGRIDZ&resolution.values=PT15M&resolution.values=PT30M&resolution.values=PT60M&dateTime.timezone=CET_CEST&dateTime.timezone_input=CET+(UTC+1)+/+CEST+(UTC+2))wrong?

[image]

haldi4803, 2024-01-22:
Not even close i'd say!
[image]

ArnieO, 2024-01-22:
> Not even close i'd say!

The price changes each hour, and I assume what you see on your bill is some kind of average over (I assume) last month. 
The price published on Entso-E (which is what you see in the user interface) is the system price, not including grid cost, fees, VAT etc. 

Example:
Your graph here above from the user interface shows 0,0778 CHF/kWh for the period 16h to 17h.
Entso-E graph for the same period says 82,26 EUR/MWh = 0,08226 EUR / kWh. 

So I find the values on the user interface are correct.
[image]

If you want to model the price better to take into account those other costs you need to use the multiplier, or try the test version here: https://github.com/UtilitechAS/amsreader-firmware/issues/252#issuecomment-1879775316