UtilitechAS/amsreader-firmware #454: Future Energy Price missing today (feb 16th 2023)

andreas1974, 2023-02-16:
The frame showing "Future Energy Price" is missing today. Is there a known reason?
Reloading the page did not help (I tried from a phone and from a computer, so it's not a local browser error)
Restarting the POW-P1 device did not help.

I am however still running ver 2.1.14 (due to an awkward placement of the meter I am somewhat reluctant to go there just to put external power, for the upgrade, as long as everything works).
Has there been any change in the Entso-e data, so I really need to upgrade to get the future prices back?

gskjold, 2023-02-16:
ENTSO-E have changed their URL. Fixed in 2.2.6 and later. 
See issue #431 
https://github.com/UtilitechAS/amsreader-firmware/releases/tag/v2.2.6

closed 2023-02-16 by gskjold

andreas1974, 2023-02-16:
I updated and got it running. Now the Entso-e price pane is back.
And I also got the update that I had previously asked for: spot price calculations on real time production. Very nice!
Thank you!