UtilitechAS/amsreader-firmware #455: SSID with special characters not working with latest release

MrAdam, 2023-02-17:
After updating to the latest release a few days ago (v2.2.8), my Pow-K+ has been unable to connect to my WiFi.
I found out that it works fine with a SSID without special characters, but my home network "Honoré" would not work anymore.

This has been working for quite some time now, so I guess it's a regression in one of the newer versions of the firmware.

ArnieO, 2023-02-17:
Which was the latest firmware where it worked?

MrAdam, 2023-02-17:
I unfortunately can't remember, but it was pre 2.2.

MrAdam, 2023-02-17:
Found it!
I restored a weekly backup of my `home-assistant` directory, and found it in the `core.device_registry` file.
It was running `v2.1.16` this sunday, when it was still working.

ArnieO, 2023-02-17:
> I unfortunately can't remember, but it was pre 2.2.

Fine - then it is likely that something happened in the change from 2.1.x to 2.2.x (a major update).

gskjold, 2023-02-17:
Sounds like my ascii filter is a bit too tight, I will have a look at that next time i look at the code

closed 2023-03-11 by gskjold