UtilitechAS/amsreader-firmware #456: 2.2.8 / ESP32 does not fetch prices

tronde-ams, 2023-02-17:
FW2.2.8 / Norway / Aidon 3p it / ESP32 / MQTT
No new prices with own api key. I see this even after re-boot.
[image]

Without key, but activated prices, I see this with error message.
[image]

Trouble at ENTSO-E or something else?

gskjold, 2023-02-17:
Seems like entso-e has some trouble. No data for any of my entso-e based applications. I suspect they have some migration issues after terminating the old endpoint on tuesday, the price publications have been unstable for the last few days

tb205gti, 2023-02-17:
Same issue here, lacking price data since yesterday. No errormessages, just no prices

tronde-ams, 2023-02-18:
Seems to be a problem at ENTSO-E, so I close this issue.

closed 2023-02-18 by tronde-ams