UtilitechAS/amsreader-firmware #462: Import and export values on top of each other

SirJMD, 2023-02-22:
**Describe the bug**
Import and export values are jumbled together.

**To Reproduce**
Have a meter with export enabled.
Same behaviour in Chrome, Edge and Internet Explorer on Win10 as well as Chrome on Android.

**Expected behavior**
Readable import and export values.

**Screenshots**
[image]

**Hardware information:**
 - Country: Denmark
 - Meter: Kamstrup
 - Encryption enabled: Yes
 - AMS reader: Pow-K+

**Relevant firmware information:**
 - Version: v2.1.17

Apart from that, AMS Reader is awesome!

gskjold, 2023-02-22:
Thank you for your feedback. The graphs have been replaced in v2.2 series firmware, so this should not be a problem anymore. There could be other issues though, so feel free to provide some feedback after testing the latest release :)

SirJMD, 2023-02-27:
> Thank you for your feedback. The graphs have been replaced in v2.2 series firmware, so this should not be a problem anymore. There could be other issues though, so feel free to provide some feedback after testing the latest release :)

I got a prompt to update prior to posting this bug report, but it seems like you have to manually update the firmware. That fixed it though :)

closed 2023-02-27 by SirJMD