UtilitechAS/amsreader-firmware #463: Wrong hourly prices from Entsoe

gudnyh, 2023-02-22:
#**Describe the bug**
Prices NO1  on display do not correspond with Day Ahead NO1 prices

**To Reproduce**
Steps to reproduce the behavior:
1. Go to display on iPad in app
2. Scroll down to prices from Entso-e
4. See prices. Not prices for selected bidding zone and not equal to prices on Fortum-app

**Expected behavior**
Prices correspond to Day Ahead prices and those shown in Fortum app. As they have been previously.
[image]
[image]
[image]

**Screenshots**
Attachment.

**Hardware information:**
 - Country: [e.g. Norway] Norway
 - Meter: [e.g. Aidon]
 - Encryption enabled [yes/no]
 - AMS reader: [e.g. Pow-U, ESP32 etc]
 - M-bus adapter (if applicable):

**Relevant firmware information:**
 - Version: [e.g. 2.1.0]
 - MQTT: [yes/no]
 - MQTT payload type: [e.g. JSON]
 - HAN GPIO: [e.g. GPIO5]
 - HAN baud and parity: [e.g. 2400 8E1]
 - Temperature sensors [e.g. 3xDS18B20]
 - ENTSO-E API enabled: [yes/no]

**Additional context**
Add any other context about the problem here.

tronde-ams, 2023-02-23:
When I choose prices in EUR x 1000 they seems to correspond with https://transparency.entsoe.eu/transmission-domain/r2/dayAheadPrices/show

but the prices in NOK seems to be too low.

gudnyh, 2023-02-23:
Prices on yesterday's attachment is  for 22.02, Wednesday. 

Target comparison is better with NordPool, this one:
https://www.nordpoolgroup.com/en/Market-data1/Dayahead/Area-Prices/ALL1/Hourly/?view=table

Prices on NordPool is pr MWH. Select NOK and /1000 to compare.

They do not correspond with those displayed on amsreader.

I take prices at NordPool as correct.

ArnieO, 2023-02-23:
I cannot see any problem. This is data from https://www.nordpoolgroup.com/en/Market-data1/Dayahead/Area-Prices/ALL1/Hourly/?view=table
[image]

When I compare with data in first post here, it seems to be identical.

gudnyh, 2023-02-23:
Ok. I see the problem now. It's about readability of the price staples view:

It is difficult for the user to match hourly prices with what date they belong to.

Since there is no date in the view, and combined with the rolling window, the user easily mixes up part of the presented set of prices that belongs to the next day, and read them as today's prices.

I did not imagine until you verified the prices that the view worked like that. 

Means somehow the view probably makes a "jump" around 13 CET to show part of next day prices. This is a bit nerdy behavior that is difficult to grasp for other than people working close to the business - as I in fact do :-D - and yet - I did not read the view as you wanted me to :-o

I suggest you put in an indication/split for date or day name - or maybe shade "next day staples" to improve readability of the price view.

gudnyh, 2023-02-23:
I close as resolved and you decide what you do with my proposals. Thanks, u-all!

closed 2023-02-26 by gskjold