UtilitechAS/amsreader-firmware #468: Clicking the configuration button doesn't load the configuration

shadowano, 2023-03-01:
**Describe the bug**
Clicking the configuration cog icon opens a small popup "Loading configuration". This is it. I see the configuration fields in the background, but the layover doesn't hide so I can't change anything. 

[image]

There are a lot of errors in the console when browsing the page
[image]

Error happening when clicking the configuration button:
[image]

Browser: Version 110.0.5481.104 (Official Build) (64-bit), also tested Edge Version 110.0.1587.57 (Official build) (64-bit)
OS: Windows 11
Same happens on Android Phone using Chrome. 

**Hardware information:**
 - Country: Norway
 - Meter: Aidon
 - Encryption enabled no
 - AMS reader: Pow-U, ESP32

**Relevant firmware information:**
 - Version: v2.2.8

gskjold, 2023-03-01:
Wild guess, maybe an illegal character in the config. Would check configuration.json

shadowano, 2023-03-03:
I downloaded the configuration file and opened in Notepad++ and there looks to be some strange characters in entsoeArea and entsoeCurrency. I haven't configured Entsoe, or I have disabled it at some point before upgrading to version 2.x

Should I change these values and reupload config? I guess I need to also get my secrets downloaded before uploading

[image]

gskjold, 2023-03-03:
Yes, fix and reupload. No need to include secrets, it will only update items included in the config file and leave all other as is.

shadowano, 2023-03-03:
thanks, that fixed it:)

closed 2023-03-03 by shadowano