UtilitechAS/amsreader-firmware #469: Home Assistant MQTT makes Home assistant crash MQTT integration

KennethLavrsen, 2023-03-01:
**Describe the bug**

Home assistant has become more and more strict with the format of MQTT sensors.
In the last 2023.3 version the amsreader HA MQTT makes HA crash the MQTT integration during boot of HA and several sensors from amsreader do not work at all any more

**To Reproduce**
Use the HA format for MQTT
Read the HA log. It is very very long full of errors

**Expected behavior**
No errors and sensors working

**Screenshots**
N/A

**Hardware information:**
 - Country: Denmark
 - Meter: Kamstrup
 - Encryption enabled yes
 - AMS reader: POW-K+
 - M-bus adapter (if applicable):

**Relevant firmware information:**
 - Version: 2.2.8
 - MQTT: [yes/no]
 - MQTT payload type: Home Assistant
 - HAN GPIO: not relevant
 - HAN baud and parity: not relevant
 - Temperature sensors No
 - ENTSO-E API enabled: yes

**Additional context**
Add any other context about the problem here.

Below are relevant logs

Missing tzzone info in time causes problem

```
2023-03-01 22:00:07.362 ERROR (MainThread) [homeassistant] Error doing job: Exception in callback MQTT._mqtt_handle_message(<paho.mqtt.cl...xffff71d5e260>)
Traceback (most recent call last):
  File "/usr/src/homeassistant/homeassistant/components/sensor/__init__.py", line 517, in state
    if value.tzinfo is None:
AttributeError: 'str' object has no attribute 'tzinfo'

The above exception was the direct cause of the following exception:

Traceback (most recent call last):
  File "/usr/local/lib/python3.10/asyncio/events.py", line 80, in _run
    self._context.run(self._callback, *self._args)
  File "/usr/src/homeassistant/homeassistant/components/mqtt/client.py", line 722, in _mqtt_handle_message
    self._mqtt_data.state_write_requests.process_write_state_requests()
  File "/usr/src/homeassistant/homeassistant/components/mqtt/models.py", line 267, in process_write_state_requests
    entity.async_write_ha_state()
  File "/usr/src/homeassistant/homeassistant/helpers/entity.py", line 559, in async_write_ha_state
    self._async_write_ha_state()
  File "/usr/src/homeassistant/homeassistant/helpers/entity.py", line 600, in _async_write_ha_state
    state = self._stringify_state(available)
  File "/usr/src/homeassistant/homeassistant/helpers/entity.py", line 565, in _stringify_state
    if (state := self.state) is None:
  File "/usr/src/homeassistant/homeassistant/components/sensor/__init__.py", line 528, in state
    raise ValueError(
ValueError: Invalid datetime: sensor.kamstrup_pricescheapest1hr has timestamp device class but provides state 2023-03-02T12:00:00Z:<class 'str'> resulting in ''str' object has no attribute 'tzinfo''
```

And further device class causes problem. Illegal combination

```
2023-03-01 21:58:56.499 WARNING (MainThread) [homeassistant.components.sensor] Entity sensor.kamstrup_hourcost (<class 'homeassistant.components.mqtt.sensor.MqttSensor'>) is using state class 'total_increasing' which is impossible considering device class ('monetary') it is using; expected None or one of 'total'; Please update your configuration if your entity is manually configured, otherwise create a bug report at https://github.com/home-assistant/core/issues?q=is%3Aopen+is%3Aissue+label%3A%22integration%3A+mqtt%22
2023-03-01 21:58:56.501 WARNING (MainThread) [homeassistant.components.sensor] Entity sensor.kamstrup_daycost (<class 'homeassistant.components.mqtt.sensor.MqttSensor'>) is using state class 'total_increasing' which is impossible considering device class ('monetary') it is using; expected None or one of 'total'; Please update your configuration if your entity is manually configured, otherwise create a bug report at https://github.com/home-assistant/core/issues?q=is%3Aopen+is%3Aissue+label%3A%22integration%3A+mqtt%22
2023-03-01 21:58:56.503 WARNING (MainThread) [homeassistant.components.sensor] Entity sensor.kamstrup_monthcost (<class 'homeassistant.components.mqtt.sensor.MqttSensor'>) is using state class 'total_increasing' which is impossible considering device class ('monetary') it is using; expected None or one of 'total'; Please update your configuration if your entity is manually configured, otherwise create a bug report at https://github.com/home-assistant/core/issues?q=is%3Aopen+is%3Aissue+label%3A%22integration%3A+mqtt%22

```

It is the tzinfo that causes MQTT integration in Home Assistant to crash and delay Home Assistant start by minutes in worst case. This is also an Home Assistant problem by not having defensive coding against errors but it does not change the fact that amrreader sends MQTT that does not comply with what Home Assistant expects

If you are a user and read this - disable the 3 sensors related to cheapest hours in Home Assistant to avoid that it crashes MQTT integration

KennethLavrsen, 2023-03-01:
I also raised bug for Home Assistant for the too aggresive code execusion https://github.com/home-assistant/core/issues/88994

LordMike, 2023-03-01:
I opened #470 for the other type mismatches HA shows in the logs. Just noticed one of them was also mentioned here.

gskjold, 2023-03-26:
The date format used is valid, specifying zulu (Z) as time zone. As far as I can tell, this have been fixed in HA 2023.3.1

closed 2023-03-26 by gskjold