UtilitechAS/amsreader-firmware #474: GUI option to change Home Assistant MQTT discovery prefix.

StarkJohan, 2023-03-07:
**Is your feature request related to a problem? Please describe.**
I manage multiple HA setups/properties with a common MQTT broker. I use separate MQTT discovery prefixes for the different properties. This separation is not possible to keep as amsreader-firmware works today.

**Describe the solution you'd like**
In addition to MQTT "Publish topic" input string in the GUI  I would like a "Discovery prefix" input string to use. This way the default "homeassistant" prefix can be overridden as easily as the "Publish topic".

**Describe alternatives you've considered**
I've built amsreader-firmware from source with my own hard coded discovery topic(s) as needed. Works fine but breaks with every upgrade.

ArnieO, 2023-03-07:
Good proposition, thank you!

gskjold, 2023-04-01:
Implemented in next patch

closed 2023-04-01 by gskjold

StarkJohan, 2023-04-01:
Great news guys!