UtilitechAS/amsreader-firmware #479: 2.2.8 upgrade to 2.2.9: external power not needed?

LordMike, 2023-03-12:
I just upgraded from 2.2.8 to 2.2.9 using the new “upgrade device” link on the dashboard. That was easy! :)

Quick question. I know from past updates that it was recommended to supply power to the device, and not run off of the meters power. So I did that. 

But the upgrade prompt said nothing about this recommendation. I kinda expected it to go to a page saying just that, but when I clicked ok it immediately began updating. 

Is external power no longer needed?

If it still is, I’ll either add a pr for that recommendation in the prompt or suggest it is added ;).

gskjold, 2023-03-12:
This depends on your device, only BUS powered devices like a Pow-U needs external power. Pow-K does not.

LordMike, 2023-03-12:
What.. really :O

So the `Pow-K+` can update on the measly power available from the meter?.. 

This issue can be closed then. I had bad info :)

closed 2023-03-13 by gskjold