UtilitechAS/amsreader-firmware #482: V 2.2.9 makes the electricity price wrong, seems to pick the price from an hour earlier

KennethLavrsen, 2023-03-12:
**Describe the bug**
A clear description of what the bug is.

**To Reproduce**
Steps to reproduce the behavior:
1. I have a token from ENTSOE and I am in Denmark DK2
2. Look at price and compare with the actual data from ENTSOE
3. Price for DK2 in DKK seems to be the price from previous hour

**Expected behavior**
I expect corrent price

**Screenshots**

**Hardware information:**
 - Country: Denmark
 - Meter: kamstrup
 - Encryption enabled yes
 - AMS reader: pow-k
 - M-bus adapter (if applicable):

**Relevant firmware information:**
 - Version: 2.2.9
 - MQTT: yes
 - MQTT payload type: not relevant. Error also seen in amsreader web interface
 - HAN GPIO: 
 - HAN baud and parity: 
 - Temperature sensors 
 - ENTSO-E API enabled: Yes

**Additional context**
The time shown in web ui is correct
But would you by any chance have a mistake in code with daylight saving time?  USA changed a week ago. EU still in wintertime. If DST code uses wrong region you could be an hour behind

KennethLavrsen, 2023-03-13:
Wanted to document the error in prices

This is the image from amrreader

[image]

And here is the table - here taken from Nordpool as it can show raw price in DKK

[image]

Totally wrong. Not even the trend is the same. These prices must be from a different region of Europe. It does not even match DK1. The number it fetches are totally wrong

I checked again here 10 hours later and the numbers are the same wrong. I also tried to reset the reader but it loaded the same wrong data. Status right now is that the price feature cannot be used for anything. You would charge your car and wash your clothers at the wrong time if you followed the numbers. I have had to remove this from Home Assistant

KennethLavrsen, 2023-03-13:
Hi. Yes. I have my own API key.

I just downgraded to 2.2.8 and now prices are correct again.

gskjold, 2023-03-13:
Interresting, no changes on price fetching with your own API key between the two versions. I will try to reproduce this though

KennethLavrsen, 2023-03-13:
I tried to upgrade to 2.2.9 again. And data is still correct. Note that I upgraded and recorded this bug report around midnigt.

Can it be that the code gets the date wrong around midnight. Say +/- 1 hour from midnight?

Could it be that the code uses a formula deciding Daylight Saving Time that assumed USA and not EU? USA changed to normal time over the weekend while EU remains on DST until 26th of March. Can it be something silly like that? If it gets the time wrong by an hour then you would pick the previous date. Just ideas.

kimsey0, 2023-03-13:
@KennethLavrsen, are you also seeing no energy usage data at 23? That does lend some credence to the daylight saving time hypothesis.
[image]

gskjold, 2023-03-14:
This is an interesting theory. The code uses CET/CEST, configured here: https://github.com/UtilitechAS/amsreader-firmware/blob/938f9f69d1d47c0db55944ffa946a2fe4342f074/lib/EntsoePriceApi/src/EntsoeApi.cpp#L20
But maybe there is something in that code that makes it not work.. Needs investigation

KennethLavrsen, 2023-03-14:
Today Tuesday the 14 Mar 2023 the 2.2.9 seems to have fetched the correct prices. We cannot rule out that the ENTSOE guys had a DST problem. 

I lost the consumption data when I downgraded so I do not have data from last hour Saturday. But I have full 24 hour data from Mondat at 23-0

KennethLavrsen, 2023-03-21:
I have not seen this again. Could have been a glitch at the source. Closing as I cannot provide the dev any means to reproduce

closed 2023-03-21 by KennethLavrsen