UtilitechAS/amsreader-firmware #488: EVN NetzNoe - T210-D

njordan77, 2023-03-17:
Hi,
i'm using an ESP8266 with the 2.29 binary to read out the T210-D via EVN/NEtzNoe (AT) smart meter. I'm online and tried to configure the HAN Hardware. Did use GPIO5 (D1) and set baud to 2400 and 8N1. Also did set the Encryption keyphrase (left authentication to 000000000, as not used). but whatever i try (also changing baud rate, ....)  simply does not make HAN green and send any data.

Is there anything else i can try. I'm using the recommended MikroElektronika M-BUS-Click Adapterboard.

Sorry and thanks for helping out.
At least from what i see in forums, this firmware should work with this smartmeter in combination of the provider.
Kind REgards
Norbert

Noschvie, 2023-03-17:
EVN NetzNoe #407

Try this : M-Bus RX auf GPIO13 / D7

njordan77, 2023-03-17:
thanks for your comments. i now tried 2400 with 8E1 and 8N1..both did not work. Which one in general is true. 2400 and 8N1???
I changed to D7 but did not help.

I have the official encryption key set and authentication key to 000000, is that the correct way?
is it important which cable Mbus+ and Mbus- is connected to which mbus-dev-board pin? up to now i did not change that and hope it does not matter....am i right??

Should HAN at least be green if the encryption code is wrong??? that is my final idea that they sent me a wrong one.....could that be the reason for not seing anything???

Finally, can I check with a voltmeter if P1 is active, like to see the 24-36V between the two pins?

Many Thanks Norbert

ArnieO, 2023-03-18:
It could be your Mbus-to-digital converter is not working as expected. A photo of your hardware might help. 
Try a Telnet debug to see whether it receives any data from your meter: https://amsleser.no/blog/post/24-telnet-debug
Then please post the output here.

njordan77, 2023-03-18:
Thanks. as written above its the recommended mbus MikroElektronika M-BUS-Click Adapterboard.

[image]

Can someone please confirm the correct serial comm parameters: is it 2400baud, 8N1 ????
Plus....what comes to Authentication Key??? I only received the encryption key?
What DC Voltage can i expect on P1 interface.....so i can check if it is active at all.

njordan77, 2023-03-18:
i now checked via debug...no new insights

`*** Remote debug - over telnet - for ESP8266 (NodeMCU) - version 3.0.5
* Host name: p1netznoe IP:192.168.1.23 Mac address:BC:DD:C2:17:5D:C0
* Free Heap RAM: 16120
* ESP SDK version: 2.2.2-dev(38a443e)
******************************************************
* Commands:
    ? or help -> display these help of commands
    q -> quit (close this connection)
    m -> display memory available
    v -> set debug level to verbose
    d -> set debug level to debug
    i -> set debug level to info
    w -> set debug level to warning
    e -> set debug level to errors
    s -> set debug silence on/off
    l -> show debug level
    t -> show time (millis)
    profiler:
      p      -> show time between actual and last message (in millis)
      p min  -> show only if time is this minimal
      P time -> set debug level to profiler
    c -> show colors
    filter:
          filter <string> -> show only debugs with this
          nofilter        -> disable the filter
    cpu80  -> ESP8266 CPU a 80MHz
    cpu160 -> ESP8266 CPU a 160MHz

* Please type the command and press enter to execute.(? or h for this help)
***
(W) (errorBlink) No HAN data received last 30s, single blink
(D) Serving /data.json over http...
(W) (errorBlink) No HAN data received last 30s, single blink
(D) Serving /data.json over http...
(W) (errorBlink) No HAN data received last 30s, single blink
(D) Serving /data.json over http...
(W) (errorBlink) No HAN data received last 30s, single blink
(D) Serving /data.json over http...
(D) Serving /data.json over http...
(W) (errorBlink) No HAN data received last 30s, single blink
(D) Serving /data.json over http...
(W) (errorBlink) No HAN data received last 30s, single blink
(D) Serving /data.json over http...
(W) (errorBlink) No HAN data received last 30s, single blink
`

njordan77, 2023-03-19:
sure. but again. can someone using NetzNOE Smartmeter confirm the communications parameters. I read 2400 and 8N1, but is this true?

Also is it GPIO13 or UART2.

As long as i have not the correct parameters its just guessing and trying various combinations that are not supposed to work.

Thanks a lot

ArnieO, 2023-03-19:
@njordan77 : This is useful information, as it shows there is no data received. 
If parity setting was wrong, you would still see incoming data. You would also see data if the problem was encryption key. 

I think you should start by debugging your M-bus interface, to ensure it outputs data. 

A connection diagram is always useful if you want assistance.

njordan77, 2023-03-19:
@ArnieO can you please confirm that this project only needs the RX part of the modbus-board to be connected to GPIO13/RXD2 of an ESP8266.
How about TX, til now i did not connect it. as its only unidirectional communications what i've seen in the WiKi.

I'm asking because other projects for ESPhome do also connect the TX part. Just to confirm i do not miss an important part.

Elsewise as you asked, i only connected the 3,3V and GND and RX to GPIO13.
Can you also confirm that if its connected to GPIO13....i have to select UART2 in the config GUI?

Thanks a lot.

njordan77, 2023-03-19:
JESUS - i'm sorry. after checking several projects and having a second thought...i found out the issue. HAN is green and data are coming in.

So embarrassing that i better not tell.... but i wrote it already above. so better tell the truth.......you need to connect (of course) TX of the MBUS board to the RX2/GPIO of the NodeMCU board.

closed 2023-03-19 by njordan77

ArnieO, 2023-03-20:
There are two types of engineers: Those who admit mistakes, and those who lie. 😅 
We will often get "blind" by our own work, this is why in cases like this I ask for a connection diagram to see if there are "obvious" mistakes.

Good to hear that you're up and running! 👍