UtilitechAS/amsreader-firmware #489: Denmark - Kamstrup not encrypted

Cvesterby, 2023-03-19:
I am using an ESP32 board with AMS Reader 2.2.9.
I get "HAN: Unknown data received, check meter config"

There is not so many options in the meter config, and just tried each of them out - to no luck.

Does anyone know what setting is should use with this version of Kamstrup?

I added a photo of numbers on my Kamstrup and also attached the data description I got from my supllier.
[image]

[teknisk beskrivelse_GB_kamstrup162-382.pdf](https://github.com/UtilitechAS/amsreader-firmware/files/11010867/teknisk.beskrivelse_GB_kamstrup162-382.pdf)

Thanks so much!

ArnieO, 2023-03-19:
Sorry to say, but the Kamstrup 382 cannot be configured to "Data push" - so this firmware cannot be used (at least not yet). 

You'll find more information in this thread: https://github.com/UtilitechAS/amsreader-firmware/issues/126

On this meter, a connected device needs to "pull" data from the meter using a Kamstrup protocol injected on the DATA-IN pin of the 6-pin connector in the expansion bay. Our current Pow-K devices do not have a layout that does not accommodate using that pin. So **if** this is implemented it will not be until the end of our current stock of Pow-K devices; maybe this summer. But we have not decided yet - so no promises.

The current firmware can be used on all Kamstrup meters marked "Kamstrup Omnipower".

Cvesterby, 2023-03-19:
Hi ArnieO

Thanks so much for taking time to answer, even tho it was not the reply I hoped for.

I will keep an eye out for updates ;)

franksemi, 2023-03-22:
Hi, im in Denmark. I have the Pow-K+ on my Kamstrup meter and it have been running without a glitch for around 6 month.
Last night at exact 23:00 it stoped getting HAN data.

Tried if I could see anything with the debug telnet, but I get the same information: "No HAN data received last 30s"

I'm running firmware v2.2.9 and I didn't change any settings.
gpk60 encryption_key and gpk61 authentication_key is still in the setting and I look trought everything else.

Anyone else experienced this and have an idea what goes on?

Cheers,
Frank
[image]
[image]

closed 2023-03-23 by gskjold