UtilitechAS/amsreader-firmware #490: IPv6 Support?, GUA/ULA?

revellion, 2023-03-20:
**Is your feature request related to a problem? Please describe.**
It's not a problem per-se, but i wonder if it's possible to add IPv6 support to the firmware?. maybe too big for esp8266 but for esp32 based it might be doable?

**Describe the solution you'd like**
For the firmware to support IPv6 in both Stateless autoconf IPv6+ND and IPv6+DHCPv6-managed

ArnieO, 2023-03-20:
Potentially an important point - sooner or later. 
I heard rumors that some Danish ISPs now only use IPv6?

revellion, 2023-03-20:
I did some quick research and it doesn't seem impossible to do.

https://github.com/espressif/arduino-esp32/tree/master/libraries/WiFi/examples/WiFiIPv6

Made a fork in an attempt to take a stab at implementing IPv6 support, but i'm fairly novice in Arduino and ESP32 work so it's a pet project at best for me.

bardahlm, 2023-03-30:
Also, it would be nice if IPv6 only is supported. For that the external services Amsreader accesses needs to be IPv6 enabled.

gskjold, 2024-01-31:
I have implemented this in our development version 2.3. Please only flash this if you are able to re-flash vanilla firmware via USB or TTL. If you are upgrading from 2.2, please back up first.

[esp32.zip](https://github.com/UtilitechAS/amsreader-firmware/files/14116712/esp32.zip)
[esp32solo.zip](https://github.com/UtilitechAS/amsreader-firmware/files/14116710/esp32solo.zip)
[esp32s2.zip](https://github.com/UtilitechAS/amsreader-firmware/files/14116711/esp32s2.zip)
[esp32c3.zip](https://github.com/UtilitechAS/amsreader-firmware/files/14116709/esp32c3.zip)

revellion, 2024-02-02:
Ooh how exciting, gonna have a stab at it when I get home from work :)

revellion, 2024-02-02:
> Be aware that IPv6 uses multicast a lot. This is sent to all nodes, possibly waking them up and draining power. See https://www.rfc-editor.org/rfc/rfc7772.txt

Would it be much worse than all the IPv4 broadcasts for ARP and the alike that can't really be filtered on?

revellion, 2024-02-11:
Uploading the 2.3 development firmware to my Pow-U to test today :)

revellion, 2024-02-11:
Hmm if I check the IPv6 box and save it doesn't seem to "stick". It just reboots with IPv6 disabled

gskjold, 2024-02-13:
thanks, I will look into that

gskjold, 2024-02-15:
I must have accidentally reverted something I shouldn't, but fixed it now. Please try this version:
[esp32s2.zip](https://github.com/UtilitechAS/amsreader-firmware/files/14292685/esp32s2.zip)
[esp32c3.zip](https://github.com/UtilitechAS/amsreader-firmware/files/14292688/esp32c3.zip)
[esp32solo.zip](https://github.com/UtilitechAS/amsreader-firmware/files/14292689/esp32solo.zip)
[esp32.zip](https://github.com/UtilitechAS/amsreader-firmware/files/14292693/esp32.zip)

revellion, 2024-02-16:
Will take the meter in today and reflash it to that :). It dropped of my network during the night so i'll have to check on it anyhow :O

revellion, 2024-02-16:
It works! :D <3

closed 2024-03-21 by gskjold