UtilitechAS/amsreader-firmware #492: Daily/Monthly usage counter grows and shrinks

roflmao, 2023-03-20:
**Describe the bug**
The daily/monthly usage mqtt senso in HA goes both up and down after some while.

**To Reproduce**
Steps to reproduce the behavior:
1. Go to 'Developer Tools'
2. Click on 'States'
3. Scroll down to '....'
4. Filter for _dayuse and _monthuse to show the entities.

**Expected behavior**
Count only upwards until day end or month end.

**Screenshots**
Daily use sensor: https://www.jottacloud.com/s/0062c97e2c0a9d04496b4feec7512fe182e
Monthly use sensor: https://www.jottacloud.com/s/00608bc3fd3daa94556a5c2b5ff9a623523

**Hardware information:**
 - Country: NO
 - Meter: Aidon
 - Encryption enabled No
 - AMS reader: Pow-U 8266
 - M-bus adapter (if applicable):

**Relevant firmware information:**
 - Version: 2.2.9
 - MQTT: yes
 - MQTT payload type: HomeAssistant
 - HAN GPIO: [e.g. GPIO5]
 - HAN baud and parity: 2400 8E1 400V TN
 - Temperature sensors: No
 - ENTSO-E API enabled: yes

gskjold, 2023-03-21:
This is a typical sign that your device is rebooting, how is the uptime on the device?

ArnieO, 2023-03-21:
Please note that the Power saving setting (https://amsleser.no/blog/post/25-pow-u-and-power-saving) does not work on ESP8266. 

If short uptime is the issue, please see User Manual chapter 5: https://amsleser.no/index.php?controller=attachment&id_attachment=38

roflmao, 2023-03-21:
Seems to have been an issue with my MQTT connection. Fixed by pointing directly to one of the three nodes in a emqx cluster.

closed 2023-03-21 by roflmao