UtilitechAS/amsreader-firmware #493: Pow-U+ does not receive any HAN data

joachimtingvold, 2023-03-22:
**Describe your problem**
My Pow-U+ does not receive any HAN data from my Kaifa meter.

**Hardware information:**
 - Country: Norway
 - Meter: Kaifa
 - Encryption enabled: no
 - AMS reader: Pow-U+
 - M-bus adapter (if applicable): [TSS721](https://www.aliexpress.com/item/32894249052.html)

**Relevant firmware information:**
 - Version: 2.2.9
 - MQTT: no
 - MQTT payload type: N/A
 - HAN GPIO: N/A
 - HAN baud and parity: 2400 8E1
 - Temperature sensors: N/A
 - ENTSO-E API enabled: no

I have a Kaifa MA304H3E meter, and have connected my Pow-U+ to it's HAN port. However, the the Pow-U+ does not receive any HAN data. Using a [MBUS-to-USB adapter](https://www.aliexpress.com/item/32894249052.html), I seem to be able to receive some data (see below). The Pow-U+ states "no HAN data received last 30s", even when debugging via Telnet.

The Pow-U+ is powered via USB-cable to USB-charger with 5V@1A. I've been able to upgrade the firmware of the Pow-U+ without issues, so I have ruled out any power issue.

The HAN cable is somewhat long (probably 6-8m), but the MBUS-to-USB adapter I'm testing with, is connected behind the same cable when testing. I've even added 2m extra patch cable when testing with the MBUS-to-USB adapter, and still receive data.

Connecting the Pow-U+ directly to the HAN port with a 15cm cable doesn't have any effect (i.e. it still says "no data last 30s"). The baud/parity (2400 8E1) is the same when using the MBUS-to-USB adapter and in Pow-U+. I've tried to change this on the Pow-U+ to 8N1 and also the "inverted" option, but still no data.

```
v
* Debug: Command received: v
* Debug level set to Verbose
(D) Serving /data.json over http...
t
* Debug: Command received: t
* Show time: On
(W t:433459ms) (errorBlink)(C0) No HAN data received last 30s, single blink
(D t:437350ms) Serving /data.json over http...
(W t:439462ms) (errorBlink)(C0) No HAN data received last 30s, single blink
(D t:442400ms) Serving /data.json over http...
(W t:445465ms) (errorBlink)(C0) No HAN data received last 30s, single blink
(D t:447467ms) Serving /data.json over http...
(W t:451467ms) (errorBlink)(C0) No HAN data received last 30s, single blink
(D t:452579ms) Serving /data.json over http...
(W t:457471ms) (errorBlink)(C0) No HAN data received last 30s, single blink
(D t:457703ms) Serving /data.json over http...
(D t:462790ms) Serving /data.json over http...
(W t:463475ms) (errorBlink)(C0) No HAN data received last 30s, single blink
(D t:467888ms) Serving /data.json over http...
(W t:469478ms) (errorBlink)(C0) No HAN data received last 30s, single blink
```

joachimtingvold, 2023-03-22:
```
Connected to: /dev/ttyUSB0
2023-03-22 03:28:47.882213: Got 82 bytes:
7E A0 27 01 02 01 10 5A 87 E6 E7 00 0F 40 00 00 00 09 0C 07 E7 03 16 03 03 1C 2A FF 80 00 00 02 01 06 00 00 08 E5 80 BB 7E 7E A0 27 01 02 01 10 5A 87 E6 E7 00 0F 40 00 00 00 09 0C 07 E7 03 16 03 03 1C 2C FF 80 00 00 02 01 06 00 00 08 E2 F2 C8 7E
2023-03-22 03:28:51.889730: Got 82 bytes:
7E A0 27 01 02 01 10 5A 87 E6 E7 00 0F 40 00 00 00 09 0C 07 E7 03 16 03 03 1C 2E FF 80 00 00 02 01 06 00 00 08 DF 2F 20 7E 7E A0 27 01 02 01 10 5A 87 E6 E7 00 0F 40 00 00 00 09 0C 07 E7 03 16 03 03 1C 30 FF 80 00 00 02 01 06 00 00 08 D9 E0 5D 7E
2023-03-22 03:28:55.891738: Got 164 bytes:
7E A0 79 01 02 01 10 80 93 E6 E7 00 0F 40 00 00 00 09 0C 07 E7 03 16 03 03 1C 32 FF 80 00 00 02 0D 09 07 4B 46 4D 5F 30 30 31 09 10 36 39 37 30 36 33 31 34 30 39 31 36 38 37 36 37 09 08 4D 41 33 30 34 48 33 45 06 00 00 08 DD 06 00 00 00 00 06 00 00 00 00 06 00 00 02 D0 06 00 00 1B AF 06 00 00 1A F0 06 00 00 10 6E 06 00 00 09 3F 06 00 00 00 00 06 00 00 09 3F 3A A2 7E 7E A0 27 01 02 01 10 5A 87 E6 E7 00 0F 40 00 00 00 09 0C 07 E7 03 16 03 03 1C 34 FF 80 00 00 02 01 06 00 00 08 DF A0 3D 7E
2023-03-22 03:28:59.893930: Got 82 bytes:
7E A0 27 01 02 01 10 5A 87 E6 E7 00 0F 40 00 00 00 09 0C 07 E7 03 16 03 03 1C 36 FF 80 00 00 02 01 06 00 00 08 E3 F4 C4 7E 7E A0 27 01 02 01 10 5A 87 E6 E7 00 0F 40 00 00 00 09 0C 07 E7 03 16 03 03 1C 38 FF 80 00 00 02 01 06 00 00 08 E9 8F 66 7E
2023-03-22 03:29:03.895572: Got 164 bytes:
7E A0 27 01 02 01 10 5A 87 E6 E7 00 0F 40 00 00 00 09 0C 07 E7 03 16 03 03 1C 3A FF 80 00 00 02 01 06 00 00 08 E6 C3 9C 7E 7E A0 79 01 02 01 10 80 93 E6 E7 00 0F 40 00 00 00 09 0C 07 E7 03 16 03 03 1D 00 FF 80 00 00 02 0D 09 07 4B 46 4D 5F 30 30 31 09 10 36 39 37 30 36 33 31 34 30 39 31 36 38 37 36 37 09 08 4D 41 33 30 34 48 33 45 06 00 00 08 E1 06 00 00 00 00 06 00 00 00 00 06 00 00 02 D8 06 00 00 1B A1 06 00 00 1A E4 06 00 00 10 81 06 00 00 09 40 06 00 00 00 00 06 00 00 09 40 67 1A 7E
2023-03-22 03:29:07.897677: Got 82 bytes:
7E A0 27 01 02 01 10 5A 87 E6 E7 00 0F 40 00 00 00 09 0C 07 E7 03 16 03 03 1D 02 FF 80 00 00 02 01 06 00 00 08 DC 33 33 7E 7E A0 27 01 02 01 10 5A 87 E6 E7 00 0F 40 00 00 00 09 0C 07 E7 03 16 03 03 1D 04 FF 80 00 00 02 01 06 00 00 08 E4 35 89 7E
2023-03-22 03:29:11.899886: Got 82 bytes:
7E A0 27 01 02 01 10 5A 87 E6 E7 00 0F 40 00 00 00 09 0C 07 E7 03 16 03 03 1D 06 FF 80 00 00 02 01 06 00 00 08 DF DE 04 7E 7E A0 27 01 02 01 10 5A 87 E6 E7 00 0F 40 00 00 00 09 0C 07 E7 03 16 03 03 1D 08 FF 80 00 00 02 01 06 00 00 08 DE 76 18 7E
2023-03-22 03:29:15.901905: Got 164 bytes:
7E A0 79 01 02 01 10 80 93 E6 E7 00 0F 40 00 00 00 09 0C 07 E7 03 16 03 03 1D 0A FF 80 00 00 02 0D 09 07 4B 46 4D 5F 30 30 31 09 10 36 39 37 30 36 33 31 34 30 39 31 36 38 37 36 37 09 08 4D 41 33 30 34 48 33 45 06 00 00 08 E3 06 00 00 00 00 06 00 00 00 00 06 00 00 02 D5 06 00 00 1B BD 06 00 00 1A F7 06 00 00 10 84 06 00 00 09 41 06 00 00 00 00 06 00 00 09 41 B3 13 7E 7E A0 27 01 02 01 10 5A 87 E6 E7 00 0F 40 00 00 00 09 0C 07 E7 03 16 03 03 1D 0C FF 80 00 00 02 01 06 00 00 08 DD 9B 2F 7E
2023-03-22 03:29:19.904082: Got 82 bytes:
7E A0 27 01 02 01 10 5A 87 E6 E7 00 0F 40 00 00 00 09 0C 07 E7 03 16 03 03 1D 0E FF 80 00 00 02 01 06 00 00 08 EB 95 79 7E 7E A0 27 01 02 01 10 5A 87 E6 E7 00 0F 40 00 00 00 09 0C 07 E7 03 16 03 03 1D 10 FF 80 00 00 02 01 06 00 00 08 E8 F7 53 7E
2023-03-22 03:29:23.905730: Got 164 bytes:
7E A0 27 01 02 01 10 5A 87 E6 E7 00 0F 40 00 00 00 09 0C 07 E7 03 16 03 03 1D 12 FF 80 00 00 02 01 06 00 00 08 E3 9F EF 7E 7E A0 79 01 02 01 10 80 93 E6 E7 00 0F 40 00 00 00 09 0C 07 E7 03 16 03 03 1D 14 FF 80 00 00 02 0D 09 07 4B 46 4D 5F 30 30 31 09 10 36 39 37 30 36 33 31 34 30 39 31 36 38 37 36 37 09 08 4D 41 33 30 34 48 33 45 06 00 00 08 DE 06 00 00 00 00 06 00 00 00 00 06 00 00 02 DA 06 00 00 1B 9B 06 00 00 1A F1 06 00 00 10 71 06 00 00 09 42 06 00 00 00 00 06 00 00 09 3F 7A FE 7E
2023-03-22 03:29:27.907435: Got 82 bytes:
7E A0 27 01 02 01 10 5A 87 E6 E7 00 0F 40 00 00 00 09 0C 07 E7 03 16 03 03 1D 16 FF 80 00 00 02 01 06 00 00 08 DB 22 57 7E 7E A0 27 01 02 01 10 5A 87 E6 E7 00 0F 40 00 00 00 09 0C 07 E7 03 16 03 03 1D 18 FF 80 00 00 02 01 06 00 00 08 E2 41 F6 7E
2023-03-22 03:29:31.909473: Got 82 bytes:
7E A0 27 01 02 01 10 5A 87 E6 E7 00 0F 40 00 00 00 09 0C 07 E7 03 16 03 03 1D 1A FF 80 00 00 02 01 06 00 00 08 DC 07 2C 7E 7E A0 27 01 02 01 10 5A 87 E6 E7 00 0F 40 00 00 00 09 0C 07 E7 03 16 03 03 1D 1C FF 80 00 00 02 01 06 00 00 08 DA FC 4E 7E
2023-03-22 03:29:35.911740: Got 164 bytes:
7E A0 79 01 02 01 10 80 93 E6 E7 00 0F 40 00 00 00 09 0C 07 E7 03 16 03 03 1D 1E FF 80 00 00 02 0D 09 07 4B 46 4D 5F 30 30 31 09 10 36 39 37 30 36 33 31 34 30 39 31 36 38 37 36 37 09 08 4D 41 33 30 34 48 33 45 06 00 00 08 DC 06 00 00 00 00 06 00 00 00 00 06 00 00 02 D7 06 00 00 1B A4 06 00 00 1A E8 06 00 00 10 60 06 00 00 09 41 06 00 00 00 00 06 00 00 09 3D 3C 06 7E 7E A0 27 01 02 01 10 5A 87 E6 E7 00 0F 40 00 00 00 09 0C 07 E7 03 16 03 03 1D 20 FF 80 00 00 02 01 06 00 00 08 EA 8D 4E 7E
2023-03-22 03:29:39.914112: Got 82 bytes:
7E A0 27 01 02 01 10 5A 87 E6 E7 00 0F 40 00 00 00 09 0C 07 E7 03 16 03 03 1D 22 FF 80 00 00 02 01 06 00 00 08 E7 D3 97 7E 7E A0 27 01 02 01 10 5A 87 E6 E7 00 0F 40 00 00 00 09 0C 07 E7 03 16 03 03 1D 24 FF 80 00 00 02 01 06 00 00 08 DF D5 2D 7E
2023-03-22 03:29:43.915932: Got 164 bytes:
[... cut ...]
  port='/dev/ttyUSB0',
  baudrate=2400,
  parity=serial.PARITY_EVEN,
  stopbits=serial.STOPBITS_ONE,
  bytesize=serial.EIGHTBITS,
  timeout=4)
print("Connected to: " + ser.portstr)

while True:
  bytes = ser.read(1024)
  date = datetime.now()
  if bytes:
    print(f"{date}: Got {len(bytes)} bytes:")
    bytes = ('%02x' % int(codecs.encode(bytes, 'hex'), 16)).upper()
    bytes = ' '.join(bytes[i:i+2] for i in range(0, len(bytes), 2))
    print(bytes)
  else:
    print(f"{date}: Got nothing")
```

ArnieO, 2023-03-22:
Hi Joachim,
Sorry about this - and I suppose you did not see this blog post: https://amsleser.no/blog/post/27-i-have-a-kaifanuri-meter-and-my-new-pow-u-does-not-work

I can see from your order that you were on the waiting list for Pow-U around new year, and probably received an unmodified card, before we realized this issue. If you are not comfortable with modifying the PCB yourself (cutting a track), please return it. Depending on your meter, it might also be necessary to modify the cable as explained. If you want, we'll send you a modified cable. 

Please contact me on post@amsleser.no - we'll get you up and running! 😃

joachimtingvold, 2023-03-22:
Haha! I searched the issues on GitHub, at least, and could not find any related stuff.

Easy fix, problem solved, everything OK (=

[image]
[image]

ArnieO, 2023-03-22:
Great! 👍 
Looks very neat.

I realize it might be confusing, but we use this Github for stuff related to the firmware - and try to take care of hardware related issues elsewhere.

closed 2023-03-22 by joachimtingvold