UtilitechAS/amsreader-firmware #494: Stoped getting HAN data at 23:00 2023-03-21

franksemi, 2023-03-22:
Hi, im in Denmark. I have the Pow-K+ on my Kamstrup meter (Kamstrup DK-8660 Omnipower Three-phase meter) and it have been running without a glitch for around 6 month.

Last night at exact 23:00 it stoped getting HAN data.

Tried if I could see anything with the debug telnet, but I get the same information: "No HAN data received last 30s"

I'm running firmware v2.2.9 now and I didn't change any settings. I was running firmware v2.2.8 when I noticed the issue. and thats why I updated. But that didn't fix the issue.

gpk60 encryption_key and gpk61 authentication_key is still in the setting and I look trought everything else.

Anyone else experienced this and have an idea what goes on?

Cheers,
Frank

[image]
[image]
[image]

ArnieO, 2023-03-23:
Hi Frank, 

Thank you for very thorough error report!
My guess is that your meter needs a reboot, as it seems to have stopped sending data.
Based on your report, I assume the blue blinking has also stopped; the blue part of the RGB LED of the Pow-K is directly connected to the incoming data signal. 

There is in principle two ways to reboot the meter:

1. Contact your grid company and request a remote reboot. 
2. Turn off the main fuse for a short while (I would do 30 seconds, just to be sure). The disadvantage is that power is cut, and several appliances might need to be reset (clocks etc).

We hear about these cases from time to time - meters that need restarts. 
It is uncommon but it happens.

closed 2023-03-23 by franksemi

ArnieO, 2023-03-23:
Good to hear your meter is now running again!

And thank you for kind words! Stability is due to a combination of very well-written code (@gskjold here) and a design that takes properly into account the power management including a huge super capacitor to ensure voltage is stable while the ESP transmits on Wifi. We have both been working on the HW and firmware things over the last 2-3 years, so it has taken time and efforts to get where we are today.