UtilitechAS/amsreader-firmware #496: Wrong future price, probably related to DST switch

cmock, 2023-03-25:
**Describe the bug**
Today, the "future energy price" (no API key) for tomorrow 23:00 is given as -157€/kWh. The prices for the hours up to that point look reasonable.

As this is the daylight saving time switch weekend, I guess this is related.

**To Reproduce**
Steps to reproduce the behavior:
Look at the Web UI

Tried rebooting and a different browser, same result.

**Screenshots**

[image]

**Hardware information:**
 - Country: Austria
 - Meter: Sagemcom T210D
 - Encryption enabled yes
 - AMS reader: ESP32
 - M-bus adapter (if applicable): resistive divider

**Relevant firmware information:**
 - Version: 2.2.9
 - MQTT: yes
 - MQTT payload type: Raw (full)
 - HAN GPIO: UART2
 - HAN baud and parity: 2400 8E1
 - Temperature sensors -
 - ENTSO-E API enabled: yes (no key)

gskjold, 2023-03-25:
Yep... If you reboot now, you will at least get rid of that rogue negative bar. Working on a fix for the clock skew that occurs in the graph.

gskjold, 2023-03-25:
I'm leaving this screenshot here as a reference for tomorrow, have to check if the price aligns after entering DST. Price zone NO2*1.25

[image]

gskjold, 2023-03-26:
Nope, this needs more work

closed 2023-03-26 by gskjold