UtilitechAS/amsreader-firmware #497: Wrong negative Amp shown

PerBuch54, 2023-03-26:
Pow P! and L+G 
When i produce solarpower the Amp shown are with a bug
I have seen up to -1.2A, when i produve more the amp for that fase goes to 100%, turns red and displays more than 2mil Amp.

gskjold, 2023-03-26:
Country?

Also, if you are able to run a debug session, that would be great: https://amsleser.no/blog/post/24-telnet-debug

PerBuch54, 2023-03-26:
It is DK1 😊

PerBuch54, 2023-03-27:
*** Remote debug - over telnet - for ESP32 - version 3.0.5
* Host name: ams-249 IP:192.168.111.75 Mac address:58:CF:79:A4:49:02
* Free Heap RAM: 120492
* ESP SDK version: v4.4.3
******************************************************
* Commands:
    ? or help -> display these help of commands
    q -> quit (close this connection)
    m -> display memory available
    v -> set debug level to verbose
    d -> set debug level to debug
    i -> set debug level to info
    w -> set debug level to warning
    e -> set debug level to errors
    s -> set debug silence on/off
    l -> show debug level
    t -> show time (millis)
    profiler:
      p      -> show time between actual and last message (in millis)
      p min  -> show only if time is this minimal
      P time -> set debug level to profiler
    c -> show colors
    filter:
          filter <string> -> show only debugs with this
          nofilter        -> disable the filter

* Please type the command and press enter to execute.(? or h for this help)
***
(D) Serving /data.json over http...
(D) Serving /data.json over http...
(V) (unwrapData)(C0) DSMR frame:
(V) 2F 4C 47 46  35 45 33 36  30 0D 0A 0D  0A 30 2D 30
(V) 3A 31 2E 30  2E 30 28 32  33 30 33 32  37 30 36 34
(V) 32 31 30 57  29 0D 0A 31  2D 30 3A 31  2E 38 2E 30
(V) 28 30 30 30  30 32 35 36  34 2E 31 38  33 2A 6B 57
(V) 68 29 0D 0A  31 2D 30 3A  32 2E 38 2E  30 28 30 30
(V) 30 30 30 30  36 37 2E 33  34 32 2A 6B  57 68 29 0D
(V) 0A 31 2D 30  3A 33 2E 38  2E 30 28 30  30 30 30 30
(V) 30 30 30 2E  32 38 34 2A  6B 56 41 72  68 29 0D 0A
(V) 31 2D 30 3A  34 2E 38 2E  30 28 30 30  30 30 32 31
(V) 33 30 2E 31  37 34 2A 6B  56 41 72 68  29 0D 0A 31
(V) 2D 30 3A 31  2E 37 2E 30  28 30 30 30  30 2E 30 30
(V) 30 2A 6B 57  29 0D 0A 31  2D 30 3A 32  2E 37 2E 30
(V) 28 30 30 30  30 2E 30 30  38 2A 6B 57  29 0D 0A 31
(V) 2D 30 3A 33  2E 37 2E 30  28 30 30 30  30 2E 30 30
(V) 30 2A 6B 56  41 72 29 0D  0A 31 2D 30  3A 34 2E 37
(V) 2E 30 28 30  30 30 30 2E  32 34 33 2A  6B 56 41 72
(V) 29 0D 0A 31  2D 30 3A 32  31 2E 37 2E  30 28 30 30
(V) 30 30 2E 30  30 30 2A 6B  57 29 0D 0A  31 2D 30 3A
(V) 32 32 2E 37  2E 30 28 30  30 30 30 2E  30 31 39 2A
(V) 6B 57 29 0D  0A 31 2D 30  3A 34 31 2E  37 2E 30 28
(V) 30 30 30 30  2E 30 30 32  2A 6B 57 29  0D 0A 31 2D
(V) 30 3A 34 32  2E 37 2E 30  28 30 30 30  30 2E 30 30
(V) 30 2A 6B 57  29 0D 0A 31  2D 30 3A 36  31 2E 37 2E
(V) 30 28 30 30  30 30 2E 30  30 38 2A 6B  57 29 0D 0A
(V) 31 2D 30 3A  36 32 2E 37  2E 30 28 30  30 30 30 2E
(V) 30 30 30 2A  6B 57 29 0D  0A 31 2D 30  3A 32 33 2E
(V) 37 2E 30 28  30 30 30 30  2E 30 30 30  2A 6B 56 41
(V) 72 29 0D 0A  31 2D 30 3A  32 34 2E 37  2E 30 28 30
(V) 30 30 30 2E  31 39 36 2A  6B 56 41 72  29 0D 0A 31
(V) 2D 30 3A 34  33 2E 37 2E  30 28 30 30  30 30 2E 30
(V) 30 30 2A 6B  56 41 72 29  0D 0A 31 2D  30 3A 34 34
(V) 2E 37 2E 30  28 30 30 30  30 2E 30 30  34 2A 6B 56
(V) 41 72 29 0D  0A 31 2D 30  3A 36 33 2E  37 2E 30 28
(V) 30 30 30 30  2E 30 30 30  2A 6B 56 41  72 29 0D 0A
(V) 31 2D 30 3A  36 34 2E 37  2E 30 28 30  30 30 30 2E
(V) 30 34 32 2A  6B 56 41 72  29 0D 0A 31  2D 30 3A 33
(V) 32 2E 37 2E  30 28 32 32  39 2E 34 2A  56 29 0D 0A
(V) 31 2D 30 3A  35 32 2E 37  2E 30 28 32  33 30 2E 35
(V) 2A 56 29 0D  0A 31 2D 30  3A 37 32 2E  37 2E 30 28
[... 95 lines cut ...]
(V) 30 30 30 2E  31 39 36 2A  6B 56 41 72  29 0D 0A 31
(V) 2D 30 3A 34  33 2E 37 2E  30 28 30 30  30 30 2E 30
(V) 30 30 2A 6B  56 41 72 29  0D 0A 31 2D  30 3A 34 34
(V) 2E 37 2E 30  28 30 30 30  30 2E 30 30  33 2A 6B 56
(V) 41 72 29 0D  0A 31 2D 30  3A 36 33 2E  37 2E 30 28
(V) 30 30 30 30  2E 30 30 30  2A 6B 56 41  72 29 0D 0A
(V) 31 2D 30 3A  36 34 2E 37  2E 30 28 30  30 30 30 2E
(V) 30 34 33 2A  6B 56 41 72  29 0D 0A 31  2D 30 3A 33
(V) 32 2E 37 2E  30 28 32 32  39 2E 34 2A  56 29 0D 0A
(V) 31 2D 30 3A  35 32 2E 37  2E 30 28 32  33 30 2E 37
(V) 2A 56 29 0D  0A 31 2D 30  3A 37 32 2E  37 2E 30 28
(V) 32 33 30 2E  31 2A 56 29  0D 0A 31 2D  30 3A 33 31
(V) 2E 37 2E 30  28 30 30 30  2E 39 2A 41  29 0D 0A 31
(V) 2D 30 3A 35  31 2E 37 2E  30 28 30 30  30 2E 30 2A
(V) 41 29 0D 0A  31 2D 30 3A  37 31 2E 37  2E 30 28 30
(V) 30 30 2E 31  2A 41 29 0D  0A 21 39 42  30 30 0D 0A
(V)
(D) (readHanPort)(C0) Received valid DSMR at 0
(V) (EnergyAccounting) Adding 0.0000 kWh import
(V) (EnergyAccounting)  and 0.0000 DKK
(V) (EnergyAccounting) Adding 0.0000 kWh export
(V) (EnergyAccounting)  and 0.0000 DKK
(V) (EnergyAccounting)  calculating threshold, currently at 0
(V) (EnergyAccounting)  new threshold 0
(D) Serving /data.json over http...

PerBuch54, 2023-03-27:
[image]

PerBuch54, 2023-03-27:
[image]

PerBuch54, 2023-03-27:
Another odd thing around 11-13 yesterday the house idle use seems to be oround 200W, in real it is almost newer under 350W,but that could bee a L+G problem. The in/out power readings seems ti match my smart meter.

PerBuch54, 2023-03-27:
It is DK1 zone

gskjold, 2023-03-27:
Thanks for the debug. I think I know what is going on here.

If I remember correctly, Pow-P1 has esp32s2 chip, double check on status page before uploading.
Here is a test firmware: [esp32s2.zip](https://github.com/UtilitechAS/amsreader-firmware/files/11075531/esp32s2.zip) Unzip and then upload firmware.bin file in the firmware section on status page:
[image]

gskjold, 2023-03-27:
Yes, only upload firmware.bin

PerBuch54, 2023-03-27:
[image]

PerBuch54, 2023-03-27:
No negative current even when i am exporting 383W ?

gskjold, 2023-03-27:
Apparently not. At least in the payloads you attached earlier, L2 current was always 0.000

gskjold, 2023-03-27:
```
/LGF5E360

0-0:1.0.0(230327064230W)
1-0:1.8.0(00002564.183*kWh)
1-0:2.8.0(00000067.342*kWh)
1-0:3.8.0(00000000.284*kVArh)
1-0:4.8.0(00002130.176*kVArh)
1-0:1.7.0(0000.002*kW)
1-0:2.7.0(0000.007*kW)
1-0:3.7.0(0000.000*kVAr)
1-0:4.7.0(0000.244*kVAr)
1-0:21.7.0(0000.000*kW)
1-0:22.7.0(0000.014*kW)
1-0:41.7.0(0000.002*kW)
1-0:42.7.0(0000.000*kW)
1-0:61.7.0(0000.008*kW)
1-0:62.7.0(0000.000*kW)
1-0:23.7.0(0000.000*kVAr)
1-0:24.7.0(0000.196*kVAr)
1-0:43.7.0(0000.000*kVAr)
1-0:44.7.0(0000.003*kVAr)
1-0:63.7.0(0000.000*kVAr)
1-0:64.7.0(0000.043*kVAr)
1-0:32.7.0(229.4*V)
1-0:52.7.0(230.7*V)
1-0:72.7.0(230.1*V)
1-0:31.7.0(000.9*A)
1-0:51.7.0(000.0*A)
1-0:71.7.0(000.1*A)
!9B00
```

PerBuch54, 2023-03-27:
The smart meter only shows W not current.

gskjold, 2023-03-30:
I'm not sure why you never get a negative current, but at least the bug for L2 crazy current value is fixed

closed 2023-03-30 by gskjold