UtilitechAS/amsreader-firmware #500: Sky high consumption.

Spexdallas, 2023-03-28:
My consumption all of a sudden is sky high, dont know why.

**Hardware information:**
 - Country: [Sweden SE3]
 - Meter: [Kamstrup 8660]
 - Encryption enabled [no]
 - AMS reader: [Pow-k esp8266]
 - M-bus adapter (if applicable):

**Relevant firmware information:**
 - Version: [2.2.11]
 - MQTT: [yes]
 - MQTT payload type: [Homeassistant]
 - HAN GPIO: [GPIO12]
 - HAN baud and parity: [2400 8N1]
 - Temperature sensors [No]
 - ENTSO-E API enabled: [
[debug.txt](https://github.com/UtilitechAS/amsreader-firmware/files/11092649/debug.txt)
[image]
no]

gskjold, 2023-03-29:
Interesting, your consumption on the graphs seems to be really high in general, is this a large installation?

Spexdallas, 2023-03-29:
This is a normal house with normal consumption i think, i will post a screenshot from my provider aswell.
[image]

ArnieO, 2023-03-29:
So there is a 10x error in the daily consumption, so apparently two issues here:

- Single hourly measurement point going crazy
- 10x error 

@gskjold : We have seen some cases before where single graph points "go crazy". Could there be a rollover data type error here?

gskjold, 2023-03-29:
Could be some rollover issue here, needs investigation. 

But also, you need a 0.1x multiplier on kwh:
[image]

Spexdallas, 2023-03-29:
Is there a way to reset the statistics in the pow-k? I reverted back to 2.2.10 and now it looks like i used to at least in Homeassistant.

gskjold, 2023-03-30:
Pretty sure I have found the reason for this. Fix will be included in next patch

closed 2023-03-30 by gskjold

Spexdallas, 2023-03-30:
Everything looks like it should now when reverted back to 2.2.10 + adding multiplier.
Will run this fw for a month and evaluate.