UtilitechAS/amsreader-firmware #501: Decryption failed NetzBurgenland

peSchilling, 2023-03-28:
**Describe your problem**
Servus
I tested it with my Grid Provider and the decryption failed.
Can it be that the data telegram is not compatible?
Grid Provider is Netz Burgenland.

**Hardware information:**
 - Country: [Austria]
 - Meter: [Landis & Gyr E450]
 - Encryption enabled [yes]
 - AMS reader: [Pow-U]
 

**Relevant firmware information:**
 - Version: [2.2.9]
 - MQTT: [yes]
 - HAN baud and parity: [2400 8N1]
 - ENTSO-E API enabled: [no]

(D) Serving /data.json over http...
(V) (unwrapData)(C0) HDLC frame:
(V) 7E A0 76 CE  FF 03 13 3C  02 E6 E7 00  DB 08 4C 47 
(V) 5A 67 72 A9  A1 11 5E 30  00 21 80 F6  DD 28 EA CE 
(V) 53 31 21 48  DF 2C 47 35  48 E4 B3 76  D8 27 8D 79 
(V) C6 74 75 71  49 14 5C 31  0C 9B 95 59  04 90 69 F4 
(V) 57 42 68 F5  E6 E3 04 A1  5B 90 1B DA  DA 13 42 5C 
(V) 50 AC BF 3A  91 C0 94 CC  D6 AE 5F 73  68 8F 11 91 
(V) 19 F3 57 AC  00 37 62 0E  E6 CB 20 4C  F0 63 4E B6 
(V) F0 15 A6 FF  AE 6B 98 7E  
(V) (unwrapData)(C0) LLC frame:
(V) E6 E7 00 DB  08 4C 47 5A  67 72 A9 A1  11 5E 30 00 
(V) 21 80 F6 DD  28 EA CE 53  31 21 48 DF  2C 47 35 48 
(V) E4 B3 76 D8  27 8D 79 C6  74 75 71 49  14 5C 31 0C 
(V) 9B 95 59 04  90 69 F4 57  42 68 F5 E6  E3 04 A1 5B 
(V) 90 1B DA DA  13 42 5C 50  AC BF 3A 91  C0 94 CC D6 
(V) AE 5F 73 68  8F 11 91 19  F3 57 AC 00  37 62 0E E6 
(V) CB 20 4C F0  63 4E B6 F0  15 A6 FF AE  
(V) (unwrapData)(C0) GCM frame:
(V) DB 08 4C 47  5A 67 72 A9  A1 11 5E 30  00 21 80 F6 
(V) DD 28 EA CE  53 31 21 48  DF 2C 47 35  48 E4 B3 76 
(V) D8 27 8D 79  C6 74 75 71  49 14 5C 31  0C 9B 95 59 
(V) 04 90 69 F4  57 42 68 F5  E6 E3 04 A1  5B 90 1B DA 
(V) DA 13 42 5C  50 AC BF 3A  91 C0 94 CC  D6 AE 5F 73 
(V) 68 8F 11 91  19 F3 57 AC  00 37 62 0E  E6 CB 20 4C 
(V) F0 63 4E B6  F0 15 A6 FF  AE 
(W) (printHanReadError)(C0) Decryption failed
(D) Serving /data.json over http...
(V) (unwrapData)(C0) HDLC frame:
(V) 7E A0 76 CE  FF 03 13 3C  02 E6 E7 00  DB 08 4C 47 
(V) 5A 67 72 A9  A1 11 5E 30  00 21 80 F7  FE B8 07 C6 
(V) 72 B1 90 AE  AC 15 D0 AD  95 7B AC 13  7E 67 D8 A2 
(V) F0 43 51 3C  63 B6 A1 89  10 AE 9A 7E  55 4A 12 49 
(V) B9 6D EB A5  7B 57 03 69  9A BF 16 5E  AD 2A 54 41 
(V) 65 5E 79 C6  95 71 92 46  A2 3F 5B 63  0D 53 96 7D 
(V) 42 52 1F A3  80 1C 00 E8  E3 A4 B3 9B  86 CB E5 2D 
(V) 2C CA B0 E2  B7 AE 4D 7E  
(V) (unwrapData)(C0) LLC frame:
(V) E6 E7 00 DB  08 4C 47 5A  67 72 A9 A1  11 5E 30 00 
(V) 21 80 F7 FE  B8 07 C6 72  B1 90 AE AC  15 D0 AD 95 
(V) 7B AC 13 7E  67 D8 A2 F0  43 51 3C 63  B6 A1 89 10 
(V) AE 9A 7E 55  4A 12 49 B9  6D EB A5 7B  57 03 69 9A 
(V) BF 16 5E AD  2A 54 41 65  5E 79 C6 95  71 92 46 A2 
(V) 3F 5B 63 0D  53 96 7D 42  52 1F A3 80  1C 00 E8 E3 
(V) A4 B3 9B 86  CB E5 2D 2C  CA B0 E2 B7  
(V) (unwrapData)(C0) GCM frame:
(V) DB 08 4C 47  5A 67 72 A9  A1 11 5E 30  00 21 80 F7 
(V) FE B8 07 C6  72 B1 90 AE  AC 15 D0 AD  95 7B AC 13 
(V) 7E 67 D8 A2  F0 43 51 3C  63 B6 A1 89  10 AE 9A 7E 
(V) 55 4A 12 49  B9 6D EB A5  7B 57 03 69  9A BF 16 5E 
(V) AD 2A 54 41  65 5E 79 C6  95 71 92 46  A2 3F 5B 63 
(V) 0D 53 96 7D  42 52 1F A3  80 1C 00 E8  E3 A4 B3 9B 
(V) 86 CB E5 2D  2C CA B0 E2  B7 
(W) (printHanReadError)(C0) Decryption failed
(D) Serving /data.json over http...

gskjold, 2023-03-29:
It could be that this is actually not supported right now, but I'm guessing this is just a small adjustment to make it work. I'm assuming you have configured your encryption key in meter configuration? And if I'm reading this data correctly, it also requires an authentication key? I could be wrong though..

Is this the older pow-u with esp8266 or pow-u+ with esp32s2? Check status page:
[image]

Note to self, look at byte 0x11 after system title, which is not 0x81 or 0x82

ArnieO, 2023-03-29:
It is the new one with ESP32-S2, shipped this month.

ArnieO, 2023-03-29:
@peSchilling : Sorry for "stupid" point, but as I have seen it happen before...:
Please double check that the encryption keys are correctly copied. If they starts with "0x" (denoting hexadecimal) you do not need to copy the "0x" part of the strings.

ArnieO, 2023-03-29:
> @ArnieO Yes I ordered this month and received it on Monday.
I know - it was me that shipped it. 😉 

Thank you for having double checked encryption keys, and confirmed you have a two-part key.
I believe we have what we need now, please just give @gskjold a bit of time to look into adapting the code. 

The documentation from Netz Burgenland on the "kundenschnittstelle" is not very detailed, this is what I find:
https://www.netzburgenland.at/kundenservice/smart-metering/smart-metering/kundenschnittstelle.html
https://www.netzburgenland.at/fileadmin/NB_pdf_NEU/Smart_Meter/Spezifikation_Kundenschnittstelle_E450_korr_2.pdf

If you have contact with other users online in Austria that have some more information, specifically regarding the data format, please share it. (Reading technical German is "kein Problem".)

@gskjold should be able to sort this out.

Noschvie, 2023-03-29:
@peSchilling are you sure that you need the "Authentication key"? I don't belive that Netz Burgenland uses a "two-part key"

gskjold, 2023-03-29:
The authentication key will only be used if the payload specifies that authentication is used. In this case (judging from the data dump in first post) it is specified by the payload that authentication is enabled.

peSchilling, 2023-03-29:
@Noschvie about the two part key .
 I have get from NetBurgenland a textfile with the two Keys
Authentication Key:
C0C1C2C3C4C5C6C7C8C9CACBCCCDCECF
Encyrption Key:
34743677397A24432646294A404E6352

and i worked with an Online Tool Tool decrypt the Data.

https://gchq.github.io/CyberChef/#recipe=AES_Encrypt(%7B'option':'Hex','string':'34743677397A24432646294A404E6352'%7D,%7B'option':'Hex','string':'4C475A6772A9A11100057E3B'%7D,'GCM','Hex','Hex',%7B'option':'Hex','string':'30C0C1C2C3C4C5C6C7C8C9CACBCCCDCECF'%7D)&input=REY4RUFBMEUxMzA3QTNGQkMxRjlFMEI1NzZDNjM4RkIzOUQ3MTkyNjk2M0E3QzY4RjQ1NDQ5ODUwOTlFQUIzRTY2RDBDRjRBRTgwQTJDMTExQzc0MEU1ODk3RDVCQ0Y4OTMwN0Y3NzFDN0RDRTBERUM4NkUzOEE5Q0IyQUY3NDRBQjM4QUZBRTYwRUM2NTA0NEYxQzY2RUM2NkIyRUQ2MzdGMDk2ODZFRkZBNEUyM0E1Mw

gskjold, 2023-03-29:
They love proprietary formats in Austria... I can make it decrypt the payload no problem, but the decrypted payload is not something we support at the moment. I will get back to you when I have looked more on the data.

Noschvie, 2023-03-29:
Ok, Encyrption Key and Authentication Key...

ArnieO, 2023-03-29:
> I am unsure of translation of these:

Google translate says
Wirkleistung bezug = active power reference
Wirkleistung rückspeisung = Active power feedback
(I assume that is Active power, import and export)

Wirkenergie bezug-gesamt = Total active energy purchased
Wirkenergie Rückspeisung-gesamt = Active energy recovery total
(I assume that is Accumulated energy, import and export)

gskjold, 2023-03-29:
Thanks for the translation, I should be able to make this work.

Updated breakdown, Im 
```
0f00057e41 // UI Frame header
0c07e60c0c010c232dff800000 // Date & time
020e // Structure with 14 items
1200ec // U1 = 236 V
1200ec // U2 = 236 V
1200ec // U3 = 236 V
120000 // I1 = 0.00 A
12002e // I2 = 0.46 A
12001a // I3 = 0.26 A
060000007d // Active import = 125 W
0600000000 // Active export = 0 W
0601a96ebd // Accumulated import = 27881.149 kWh
0600001dc3 // Accumulated export = 7.619 kWh
120190 // 400 ?
120003 // 3 ?
120120 // 288 ?
09083330313337313831 // Meter ID = 30137181
```

ArnieO, 2023-03-29:
> Im not sure what the last fields are, any suggestions?

I can email info.smartmeter@netzburgenland.at and ask.

gskjold, 2023-03-30:
Here comes a preview firmware with support for this format: 
[esp32s2.zip](https://github.com/UtilitechAS/amsreader-firmware/files/11108755/esp32s2.zip)

Unzip and upload firmware.bin file here:
[image]

peSchilling, 2023-03-30:
@gskjold Thanks it works .

[image]

Do you need somthing from me ? (log etc.)

peSchilling, 2023-03-30:
OK perfect

ArnieO, 2023-03-30:
Great @peSchilling ! 

@gskjold : Will this be included future firmware revisions?

closed 2023-03-30 by gskjold

ArnieO, 2024-04-22:
> @peSchilling are you sure that you need the "Authentication key"? I don't belive that Netz Burgenland uses a "two-part key"

@Noschvie : We received a question from a customer that is with Netz Niederosterreich, who just placed an order for a Pow-U, and just to be sure I answer him correctly: 
He has activated the port of his Sagemcom T-210 meter, and received one single key. 
- Is that all he needs to decrypt his meter? 
- And he enters it in the first field, with no Authentication key?
[image]

Noschvie, 2024-04-22:
@ArnieO : yes, that's all and should work.

ojessie, 2024-04-22:
I'm the one customer from NÖ ;-) Thanks for your support ... looking forward to the delivery.