UtilitechAS/amsreader-firmware #502: Wrong state class for energy dashboard in Home Assistant

LukasErichsen, 2023-03-28:
**Describe your problem**
After updating to 2.2.11 from 2.2.9, I've had a problem in HomeAssistant
[image]

"Unexpected state class
The following entities do not have the expected state class:
sensor.ams_cf58_houruse"

**Hardware information:**
 - Country: Denmark
 - Meter: Kamstrup
 - Encryption enabled: Yes
 - AMS reader: Pow-K

**Relevant firmware information:**
 - Version: [e.g. 2.2.11]
 - MQTT: Yes
 - MQTT payload type: HomeAssistant
 - HAN GPIO: Unsure
 - HAN baud and parity:  2400 8E1
 - Temperature sensors: none
 - ENTSO-E API enabled: no

gskjold, 2023-03-29:
Hmm, I did not get this issue on my test rig. But yes, I see now that it was changed from "total_increasing" to blank, which was not intentional, I will fix this in near future

fnorlund, 2023-03-30:
I had the same issue but it could be resolved by changing to "accumulated active import"