UtilitechAS/amsreader-firmware #504: Feature request: Tooltip on Up time

rogerthn2019, 2023-03-29:
**Is your feature request related to a problem? Please describe.**
Not a problem but it would be nice to see boot time.

**Describe the solution you'd like**
Tooltip YYYY-MM-DD HH:MM:SS on
[image]

gskjold, 2023-03-29:
Nice, I like the idea

ArnieO, 2023-03-29:
Potentially quite useful for debugging!
Maybe better than tooltip would be to add boot time here?
[image]

gskjold, 2023-03-30:
> Maybe better than tooltip would be to add boot time here?

Agree on this one

closed 2023-03-30 by gskjold