UtilitechAS/amsreader-firmware #505: Homeassistant integration

Kilill, 2023-03-30:
**No sensor data published to homeassistant**
finally got around to upgrade to 2.2.11 and changed mqtt payload format to Homeassistant, but no data is sent to the homeassistant/sensor topic it does get sent to the topic set in mqtt _Publish topic_

I did the typical dance of resetting the payload to anything else, rebooting, setting it back to Homeassistant to no effect. 
So any hints would be appreciated.

**Hardware information:**
 - Country: Norway
 - Meter: Kampstrup
 - Encryption enabled no
 - AMS reader: Pow-K+, ESP32

**Relevant firmware information:**
 - Version: 2.211
 - MQTT: yes
 - MQTT payload type: Homeassistant
 - HAN GPIO: NA
 - HAN baud and parity: NA
 - Temperature sensors No
 - ENTSO-E API enabled: yes

davralin, 2023-04-04:
I had the same issue here on 2.2.12.

I had to manually set `Discovery topic prefix` to `homeassistant` - even though that's supposedly default (greyed out in the background).

gskjold, 2023-04-05:
Sounds really weird, I'm not sure what to make of this yet, but will do some testing

HansHagglund, 2023-04-05:
> I had the same issue here on 2.2.12.
> 
> I had to manually set `Discovery topic prefix` to `homeassistant` - even though that's supposedly default (greyed out in the background).

Thanks! 
I had the same issue with 2.2.12 , but to manually set "discovery topic prefix" to "homeassistant" got things working again...
Almost. 
I'm not a programmer at all..so.. yeah..  but I think "Active Import" is missing in the auto-discovery "homeassistant/sensor/" -thing.
My Home assistant can't find it and I can't see it with "Mqtt Explorer" either. 
The value itself is in the amsreader/data flow.

I have restarted both Pow-K and Home Assistant several times.
(I have 24 sensors from my Pow-K with 2.2.12, in Sweden. Most of the time I am very happy with it )

[image]

davidolrik, 2023-04-06:
I'm also missing `Active Import` and a few other sensors in my home assistant payload on v2.2.12

rene-bayer, 2023-04-07:
I‘m missing Active Import and Active Export (in Watts) within my homeassistant sensors :-(

ArnieO, 2023-04-08:
To those of you who are missing sensors in HA: 
1) There has been some changes in HA lately, that might impact this. Make sure your HA installation is up to date.
2) Please check the content of the MQTT messages that are sent from the device.

rene-bayer, 2023-04-08:
Hi @ArnieO 
Both on the latest version.
There aren't pow_u_p and pow_u_po readings anymore
[image]

rene-bayer, 2023-04-08:
2.1.10 
[image]

ArnieO, 2023-04-09:
OK, we'll have to wait for @gskjold to take a look at this.

revellion, 2023-04-09:
Another rapport of the same issue. Lost active import as well

HansHagglund, 2023-04-09:
> To those of you who are missing sensors in HA:
> 
>     1. There has been some changes in HA lately, that might impact this. Make sure your HA installation is up to date.
> 
>     2. Please check the content of the MQTT messages that are sent from the device.

Hi, sorry for the delay.
1. The versions I'm running
 Home Assistant 2023.3.6 Supervisor 2023.04.0 Operating System 9.5 Frontend 20230309.1.
 
2. I posted all info I get in previous post. (I now see that there are more items missing that "p". Like temperature.

Is it possible to "downgrade" ?

rene-bayer, 2023-04-10:
Yes, Just upload the 2.2.10 bin file for your device

gskjold, 2023-04-10:
Found out what it is, fix coming up real soon. Sorry for the delay, had a bit of a Easter holiday going on

closed 2023-04-10 by gskjold

Kilill, 2023-05-18:
Just a heads up!, if you have access control on the mqtt server (as i do) make sure that the the  mqtt user your are using to publish,  _actually_ have write access to the _**homeassistant/sensor**_ topic, this bit for more than I'm willing to admit :-P