UtilitechAS/amsreader-firmware #507: Improved boot reason verbose indications on status page

ArnieO, 2023-03-31:
The boot verbose indications on the status page seem to not correspond to what is proposed in this link, especially for case 1: https://espressif-docs.readthedocs-hosted.com/projects/arduino-esp32/en/latest/api/reset_reason.html

I have a Pow-K+ operational now that reboots regularly (suspected ESP32 module error), this is what it reports:
[image]

According to the above link, case 1 is "Vbat power on reset", which I assume in ESP-IDF corresponds to ESP_RST_POWERON of the enum esp_reset_reason_t, see https://docs.espressif.com/projects/esp-idf/en/latest/esp32s2/api-reference/system/misc_system_api.html?highlight=esp_reset_reason_t#_CPPv418esp_reset_reason_t

If so, case 1 is a power-up event. The Pow-K+ has a large super capacitor, so a power-up reset during normal operation is highly unexpected. As there is a separate boot flag for brownout (case 15) , my understanding is that case 1 is a more severe power event than a brownout. Which simply does not make much sense in this case, strengthening my hypothesis that the ESP32 module on this board is indeed faulty. (An identical board with identical firmware in the same meter is dead stable.)

If I do a reset via GUI, it reports "3", which is as expected:
[image]

I believe the verbose indication for boot case 1 should be updated to indicate its severity. 

Verbose indications on the status page could be truncated versions of this (from the above link):
[image]

gskjold, 2023-03-31:
That would mean that the initial boot would look severe, how do we deal with that?

Please also review my other descriptive texts. Full list here:
https://github.com/UtilitechAS/amsreader-firmware/blob/master/lib/SvelteUi/app/src/lib/Helpers.js#L190-L222

gskjold, 2023-03-31:
> * Not sure how to distinguish cases 9 and 16 with short verbose, they are almost the same.

I think the most important thing is that the number is also visible in GUI, so that we can determine what it really is. These labels are mostly for end-user.

Updating descriptions for next patch.

closed 2023-03-31 by gskjold