UtilitechAS/amsreader-firmware #511: Real time calculation Export month wrong value for the price

PerBuch54, 2023-04-03:
Real time calculation Export month wrong value for the price, the kWh looks right but thr price seems to show the day value.

ArnieO, 2023-04-03:
Can someone else here with solar panel or wind turbine check if they see the same?

PerBuch54, 2023-04-04:
now it shows Month 3.9 kWh 1.0 DKK, that is wrong the kWh price has not ben down to 0,25 DKK

gskjold, 2023-04-04:
This is a precision error. Cost/income per month is stored without decimals, so the precision is lost. Need to increase the range and precision of these values

PerBuch54, 2023-04-04:
It is counting up now, seems like the first 2 days of the month did not get included.
The month stored without decimals are ok.

PerBuch54, 2023-04-05:
[image]

It is here, clear to se that the month are wrong.

PerBuch54, 2023-04-05:
[image]
 It is here, clear to se that the month value are wrong.

PerBuch54, 2023-05-02:
today before production starts Month 1.2 kWh 0.0 DKK
yesterday the day production had a DKK value. Seems like the first  two days of the month are sat to zero.

PerBuch54, 2023-05-02:
Seems like an error in Energy use last month (kWh), the last two days say 10 and 8 KWh, less than half the real value.

gskjold, 2023-05-02:
This is interesting, you are the second person to report such a problem. This obviously needs some investigation

ArnieO, 2023-05-02:
> Seems like an error in Energy use last month (kWh), the last two days say 10 and 8 KWh, less than half the real value.

Did you have any (known) Wifi issues in the period? 
Do you have "Auto reboot on connection problem" activated?

gskjold, 2023-05-02:
Checking my own data, it seems like I have a similar problem, will open new issue on this

PerBuch54, 2023-05-02:
ArnieO no wifi problems (i am living off making wifi) "Auto reboot on connection problem" not activated.

closed 2023-05-04 by gskjold