UtilitechAS/amsreader-firmware #515: check sum error

toast1234, 2023-04-04:
Hi, 

Lately I have been getting more and more checksum errors. Normally it happens a couple times every minute or so.
Maybe it is an obvious HW failure or is it something else? I have upgraded to v2.2.12 but it exists in earlier release as well.
I dont know If Im imagine but it feels like the meter is spitting out the data more rapidly now then before (once every second).

Sometimes it gets in to a state where there is no data received for more than 30 sec as in the second printout below. Maybe two different issues. Feels like garbage data is getting in as parts. So maybe more HW related than SW...

ESP32 - UART2 1152008N1 inverted

`(D) (readHanPort)(C1) Received valid DSMR at 0
(V) (EnergyAccounting) Adding 0.0010 kWh import
(V) (EnergyAccounting)  and 0.0000 SEK
(V) (EnergyAccounting)  calculating threshold, currently at 0
(V) (EnergyAccounting)  new threshold 0
(V) (unwrapData)(C1) DSMR frame:
(V) 2F 45 4C 4C  35 5C 32 35  33 38 33 33  36 33 35 5F 
(V) 41 0D 0A 0D  0A 30 2D 30  3A 31 2E 30  2E 30 28 32 
(V) 33 30 34 30  34 31 38 35  31 32 38 57  29 0D 0A 31 
(V) 2D 30 3A 31  2E 38 2E 30  28 30 30 30  32 36 38 34 
(V) 39 2E 35 35  31 2A 6B 57  68 29 0D 0A  31 2D 30 3A 
(V) 32 2E 38 2E  30 28 30 30  30 30 30 30  30 30 2E 30 
(V) 30 31 2A 6B  57 68 29 0D  0A 31 2D 30  3A 33 2E 38 
(V) 2E 30 28 30  30 30 30 30  30 34 36 2E  33 33 35 2A 
(V) 6B 76 61 72  68 29 0D 0A  31 2D 30 3A  34 2E 38 2E 
(V) 30 28 30 30  30 30 37 30  34 30 2E 36  35 37 2A 6B 
(V) 76 61 72 68  29 0D 0A 31  2D 30 3A 31  2E 37 2E 30 
(V) 28 30 30 30  33 2E 36 35  31 2A 6B 57  29 0D 0A 31 
(V) 2D 30 3A 32  2E 37 2E 30  28 30 30 30  30 2E 30 30 
(V) 30 2A 6B 57  29 0D 0A 31  2D 30 3A 33  2E 37 2E 30 
(V) 28 30 30 30  30 2E 30 34  39 2A 6B 76  61 72 29 0D 
(V) 0A 31 2D 30  3A 34 2E 37  2E 30 28 30  30 30 30 2E 
(V) 35 31 36 2A  6B 76 61 72  29 0D 0A 31  2D 30 3A 32 
(V) 31 2E 37 2E  30 28 30 30  30 31 2E 35  32 37 2A 6B 
(V) 57 29 0D 0A  31 2D 30 3A  34 31 2E 37  2E 30 28 30 
(V) 30 30 30 2E  37 37 32 2A  6B 57 29 0D  0A 31 2D 30 
(V) 3A 36 31 2E  37 2E 30 28  30 30 30 31  2E 33 35 31 
(V) 2A 6B 57 29  0D 0A 31 2D  30 3A 32 32  2E 37 2E 30 
(V) 28 30 30 30  30 2E 30 30  30 2A 6B 57  29 0D 0A 31 
(V) 2D 30 3A 34  32 2E 37 2E  30 28 30 30  30 30 2E 30 
(V) 30 30 2A 6B  57 29 0D 0A  31 2D 30 3A  36 32 2E 37 
(V) 2E 30 28 30  30 30 30 2E  30 30 30 2A  6B 57 29 0D 
(V) 0A 31 2D 30  3A 32 33 2E  37 2E 30 28  30 30 30 30 
(V) 2E 30 30 30  2A 6B 76 61  72 29 0D 0A  31 2D 30 3A 
(V) 34 33 2E 37  2E 30 28 30  30 30 30 2E  30 30 30 2A 
(V) 6B 76 61 72  29 0D 0A 31  2D 30 3A 36  33 2E 37 2E 
(V) 30 28 30 30  30 30 2E 30  34 39 2A 6B  76 61 72 29 
(V) 0D 0A 31 2D  30 3A 32 34  2E 37 2E 30  28 30 30 30 
(V) 30 2E 31 35  30 2A 6B 76  61 72 29 0D  0A 31 2D 30 
(V) 3A 34 34 2E  37 2E 30 28  30 30 30 30  2E 33 36 35 
(V) 2A 6B 76 61  72 29 0D 0A  31 2D 30 3A  36 34 2E 37 
(V) 2E 30 28 30  30 30 30 2E  30 30 30 2A  6B 76 61 72 
(V) 29 0D 0A 31  2D 30 3A 33  32 2E 37 2E  30 28 32 33 
(V) 35 2E 39 2A  56 29 0D 0A  31 2D 30 3A  35 32 2E 37 
(V) 2E 30 28 32  33 33 2E 35  2A 56 29 0D  0A 31 2D 30 
(V) 3A 37 32 2E  37 2E 30 28  32 33 37 2E  30 2A 56 29 
(V) 0D 0A 31 2D  30 3A 33 31  2E 37 2E 30  28 30 30 36 
(V) 2E 35 2A 41  29 0D 0A 31  2D 30 3A 35  31 2E 37 2E 
(V) 30 28 30 30  33 2E 38 2A  41 29 0D 0A  31 2D 30 3A 
(V) 37 31 2E 37  2E 30 28 30  30 35 2E 37  2A 41 29 0D 
(V) 0A 21 46 43  30 34 0D 0A  
(D) (readHanPort)(C1) Received valid DSMR at 0
(V) (EnergyAccounting) Adding 0.0010 kWh import
(V) (EnergyAccounting)  and 0.0000 SEK
(V) (EnergyAccounting)  calculating threshold, currently at 0
(V) (EnergyAccounting)  new threshold 0
(D) Serving /data.json over http...
(D) Expected auth: YWRtaW46cnh3LTAxMQ==
(D) Provided auth: YWRtaW46cnh3LTAxMQ==
(V) (unwrapData)(C1) DSMR frame:
[... 372 lines cut ...]
(V) 72 29 0D 0A  31 2D 30 3A  33 32 2E 37  2E 30 28 32 
(V) 33 34 2E 31  2A 56 29 0D  0A 31 2D 30  3A 35 32 2E 
(V) 37 2E 30 28  32 33 34 2E  33 2A 56 29  0D 0A 31 2D 
(V) 30 3A 37 32  2E 37 2E 30  28 32 33 37  2E 33 2A 56 
(V) 29 0D 0A 31  2D 30 3A 33  31 2E 37 2E  30 28 30 30 
(V) 36 2E 32 2A  41 29 0D 0A  31 2D 30 3A  35 31 2E 37 
(V) 2E 30 28 30  30 33 2E 38  2A 41 29 0D  0A 31 2D 30 
(V) 3A 37 31 2E  37 2E 30 28  30 30 35 2E  37 2A 41 29 
(V) 0D 0A 21 36  38 39 38 0D  0A 2F 45 4C  4C 35 5C 32 
(V) 35 33 38 33  33 36 33 35  5F 41 0D 0A  0D 0A 30 2D 
(V) 30 3A 31 2E  30 2E 30 28  32 33 30 34  30 34 31 39 
(V) 30 32 34 31  57 29 0D 0A  31 2D 30 3A  31 2E 38 2E 
(V) 30 28 30 30  30 32 36 38  35 30 2E 32  38 33 2A 6B 
(V) 57 68 29 0D  0A 31 2D 30  3A 32 2E 38  2E 30 28 30 
(V) 30 30 30 30  30 30 30 2E  30 30 31 2A  6B 57 68 29 
(V) 0D 0A 31 2D  30 3A 33 2E  38 2E 30 28  30 30 30 30 
(V) 30 30 34 36  2E 33 33 35  2A 6B 76 61  72 68 29 0D 
(V) 0A 31 2D 30  3A 34 2E 38  2E 30 28 30  30 30 30 37 
(V) 30 34 30 2E  37 33 35 2A  6B 76 61 72  68 29 0D 0A 
(V) 31 2D 30 3A  31 2E 37 2E  30 28 30 30  30 33 2E 35 
(V) 
(D) Serving /data.json over http...
(D) Expected auth: YWRtaW46cnh3LTAxMQ==
(D) Provided auth: YWRtaW46cnh3LTAxMQ==
(E) (unwrapData)(C1) Ended up in default case while unwrapping...(tag 39)`

gskjold, 2023-04-05:
I would start by checking all wiring and reconnecting/reseating everything. But it could also be that the size of the payload together with data rate is causing some buffer overflow on the UART. This can be adjusted in firmware, so when connections have been ruled out, we can try adjusting it.

ArnieO, 2023-04-05:
@toast1234 
Since you are at 115 200 baud, I assume you are on a meter with "Northern Europe P1" interface(I use that name because an M-bus interface is called P1 in mid Europe). 

The data output of this interface has an open collector output which on some meter needs a relatively strong pullup. If you pull to 3,3V (to fit an ESP GPIO) your pullup resistor should not be smaller than 3k. To weak pullup can cause unreliable reading.

ArnieO, 2023-04-06:
Our experience too is that the Sagemcom seems less picky on strong pullup. 
If you would share photo and/or schematics it could be easier to assist.

Based on the observations you describe, I would try to add significantly more low-ESR decoupling (ceramic capacitors is a good choice; as much as practically possible) on 3,3V as close as possible to the ESP. 

This application exploits Wifi quite heavily, and each time you refresh the GUI you are "pulling" a new Wifi transmission.

toast1234, 2023-04-07:
Thanks for the feed back. I think that for the moment I will settle with that it seems to work if I stay away from the GUI. Which is fine since I use Domoticz. If problems reoccur later on I will take your suggestions and see if I can make some improvment with the HW.
[image]

gskjold, 2023-04-07:
> if I open the web page and click around it starts to fail more easily

This is not unlikely, as navigating the GUI will spend processing time, leaving less time to empty the serial buffer.

Attached is a firmware with increased serial port debugging. It will also report HAN error to GUI, but is most likely more visible in console. I have not yet changed serial rx buffer, but we can try that next.

[esp32.zip](https://github.com/UtilitechAS/amsreader-firmware/files/11176590/esp32.zip)

toast1234, 2023-04-07:
The GUI information cycles between HAN: Serial buffer full
 and HAN: Unknown data received, check meter config

gskjold, 2023-04-07:
That worked out nicely, here is a version with fixed 768 byte buffer (instead of default 64). Will possibly make that either dynamic or configurable for next release.

[esp32.zip](https://github.com/UtilitechAS/amsreader-firmware/files/11176758/esp32.zip)

toast1234, 2023-04-07:
I still see the issue (Installed version: 088e564). Maybe some delay after boot and recovery cycle when the buffer overflow? However the overflow is only seen at boot and after that it shows HAN: Unknown data received, check meter config
. When I do get the HAN port working it seems more stable now. If however I reload the page and gets buffer overflow it is followed by Unknown data. So maybe the trick is to reset the port to recover after x number corrupt packages or after a minute or so when things settle? Or a combination of when the load is low and packages are corrupt then do a reset of port.

gskjold, 2023-04-15:
I'm adding some stuff to next patch that I think will help