UtilitechAS/amsreader-firmware #517: Upgraded from 2.1.17 to 2.2.12 now bad mqtt data

ACrazyConcept, 2023-04-06:
**Describe the bug**
After I upgraded to 2.2.12 it did seem to work for a few minutes. Then it started getting "bad" data and the web interface is sometimes not responsive.
I then tried to unplug and plug it in again and it did not get connected at all for a while.
I have now managed to get response and retrieve some debugging data.

**To Reproduce**
Steps to reproduce the behavior:
1. Go to '...'
2. Click on '....'
3. Scroll down to '....'
4. See error

**Expected behavior**
A clear description of what you expected to happen.

**Screenshots**
If applicable, add screenshots to help explain your problem.

**Hardware information:**
 - Country: Denmark
 - Meter: Kamstrup
 - Encryption enabled: Not sure what this means
 - AMS reader: Pow-K+ v1.51 July 2022 (came with firmware 2.1.17 pre-flashed)
 - M-bus adapter (if applicable):

**Relevant firmware information:**
 - Version: 2.2.12
 - MQTT: yes
 - MQTT payload type: JSON
 - HAN GPIO:
 - HAN baud and parity: 
 - Temperature sensors [e.g. 3xDS18B20]
 - ENTSO-E API enabled: [yes/no]

**Additional context**
Add any other context about the problem here.

[image]

```
*** Remote debug - over telnet - for ESP32 - version 3.0.5
* Host name: ams-cf58 IP:10.0.0.238 Mac address:58:CF:79:A3:FE:88
* Free Heap RAM: 119132
* ESP SDK version: v4.4.3
******************************************************
* Commands:
    ? or help -> display these help of commands
    q -> quit (close this connection)
    m -> display memory available
    v -> set debug level to verbose
    d -> set debug level to debug
    i -> set debug level to info
    w -> set debug level to warning
    e -> set debug level to errors
    s -> set debug silence on/off
    l -> show debug level
    t -> show time (millis)
    profiler:
      p      -> show time between actual and last message (in millis)
      p min  -> show only if time is this minimal
      P time -> set debug level to profiler
    c -> show colors
    filter:
          filter <string> -> show only debugs with this
          nofilter        -> disable the filter

* Please type the command and press enter to execute.(? or h for this help)
***
(D) Serving /data.json over http...
(D) Serving /data.json over http...
[... 397 lines cut ...]
(V) 02 25 0A 0E  4B 61 6D 73  74 72 75 70  5F 56 30 30
(V) 30 31 09 06  01 01 01 08  00 FF 06 00  14 A5 26 09
(V) 06 01 01 02  08 00 FF 06  00 00 00 00  09 06 01 01
(V) 03 08 00 FF  06 00 00 30  E1 09 06 01  01 04 08 00
(V) FF 06 00 08  E4 E4 09 06  01 01 00 00  01 FF 06 01
(V) 4E 99 59 09  06 01 01 01  07 00 FF 06  00 00 01 DE
(V) 09 06 01 01  02 07 00 FF  06 00 00 00  00 09 06 01
(V) 01 03 07 00  FF 06 00 00  00 00 09 06  01 01 04 07
(V) 00 FF 06 00  00 00 AD 09  06 00 01 01  00 00 FF 09
(V) 0C 07 E7 04  06 04 0D 32  14 FF 80 00  80 09 06 01
(V) 01 20 07 00  FF 12 00 E8  09 06 01 01  1F 07 00 FF
(V) 06 00 00 00  F1 09 06 01  01 15 07 00  FF 06 00 00
(V) 01 DE 09 06  01 01 21 07  00 FF 12 00  5E 09 06 01
(V) 01 0D 07 00  FF 12 00 5E  09 06 01 01  16 07 00 FF
(V) 06 00 00 00  00 09 06 01  01 16 08 00  FF 06 00 00
(V) 00 00 09 06  01 01 15 08  00 FF 06 00  14 A5 26
(V) (readHanPort)(C0) DLMS
(V) (EnergyAccounting) Adding 0.0013 kWh import
(V) (EnergyAccounting)  calculating threshold, currently at 0
(V) (EnergyAccounting)  new threshold 0
(D) Serving /data.json over http...
(D) Serving /configuration.json over http...
(D) Serving /data.json over http...
(D) Handling save method from httpReceived meter configReceived WiFi configReceived MQTT configReceived web configReceived Debug configReceived price A
```

ArnieO, 2023-04-06:
Thank you for your thorough bug report. However, I see nothing wrong in what you're sending. 
Can you elaborate on what kind of bad data you receive on MQTT?

Your difficulties to connect indicates a potential issue in your Wifi network? I always recommend to set a static IP address, could you try that and see if things improve?

Regarding:
[image]
Your HAN port is encrypted, so to receive data at all you should have entered encryption keys that you received when activating the HAN port. I assume you remember this now. :wink:

There has been some reports on difficulties in upgrading from v2.1.x to v2.2.x, it was a quite substantial upgrade.

Try the following:

1. Save your current configuration to a file (info screen / Backup & restore)
2. Do a Factory reset (config screen, red button on the bottom). You will then lose all historical values as well as current configuration setup, but that can be restored from the file.
3. Your device will now be in Access Point mode (yellow blinks on the LED)
4. Connect to the Access point as you did on initial config when your device was new, connect it to your Wifi (and as indicated: preferably provide a static IP address for it).
5. After reboot, you can upload the config file you saved earlier.

ACrazyConcept, 2023-04-06:
Thanks for the swift feedback.

Regarding the bad data I posted a screenshot. But should of course also have included how it looks usually.
As can be seen below there are many more attributes.
As for wifi the connection was rock solid on 2.1.17. I am using ip reservation. Just tried set a static ip now but with no improvement.
Yes (of course :)) I have encryption key and authentication key.
So I will attempt a reset as per your suggestion.
[image]

ArnieO, 2023-04-06:
Hm... OK, a lot of sensors missing. It could look as if the MQTT payload setting is incorrect (did not get transferred correctly from v2.1.x)? 
Awaiting to see if a Factory Reset sorts this out.

ACrazyConcept, 2023-04-06:
I did the reset and after uploading the saved config and rebooting it still behaves the same.
Could it be some issue with mqtt?

I see this in the mosquitto log.

```
1680787925: New connection from 10.0.0.238:61409 on port 1883.
1680787925: New client connected from 10.0.0.238:61409 as 29880569 (p2, c1, k10, u'hass').
1680788248: Client 29880569 has exceeded timeout, disconnecting.
1680788723: New connection from 10.0.0.238:58119 on port 1883.
1680788723: New client connected from 10.0.0.238:58119 as 29880569 (p2, c1, k10, u'hass').
```

ArnieO, 2023-04-06:
Can you show me a screendump of your config page?

ACrazyConcept, 2023-04-06:
Sure. I am not very good with mqtt to be honest. It was just working for such a long period I am not sure it is setup properly. There could easily be some weird stuff.

[image]

ArnieO, 2023-04-06:
Yes, might be something related to MQTT - but I think there should be a GUI warning if there are MQTT issues (like connection problems). I see some points you might consider adjusting:

- Select Price region and currency, and enable price fetch
[image]
- You use local DNS, are you sure 10.0.0.20 is a good choice here? I usually specify a public DNS, and 1.1.1.1 works fine:
[image]
- MQTT 1: This could be the reason for the missing sensors: I think you want to select Home Assistant payload here? 
[image]

- MQTT 2: Do you have spaces or special characters in MQTT username and/or password? I believe there has been some issues there. You might want to try to change your Mosquitto to no user/password and try without  - just to see if that sorts things out. 

I still don't understand the issues you have regarding connection / UI reactivity, but try these adjustments first and see what it gives.

ACrazyConcept, 2023-04-06:
Thanks!
I just tried the HomeAssistant payload type. It seems to be working without any issues. Will let it run for a while.

ACrazyConcept, 2023-04-07:
I think I am good with using the HomeAssistant payload type. Thanks again 👌

closed 2023-04-07 by ACrazyConcept