UtilitechAS/amsreader-firmware #523: Rebooting every 2-3 hours after upgrade to 2.2.12 (or 2.2.13)

bmork, 2023-04-12:
I noticed semi-frequent reboots after upgrading from 2.2.9 to 2.2.12.  Sometimes only after a few minutes uptime. And never more than a few hours apart.

Debug logs over telnet showed absolutely nothing unusual.  The telnet session would just terminate on reboot, as expected I guess, without any useful output. I have not tried serial debugging.

Downgrading to 2.2.11 fixed the issue. It seemed as stable as 2.2.9 had been, although only tested for a few days.

Since 2.2.13 was released while I was testing this, I tried that verson as well.  With the same instability as 2.2.12.

**Hardware information:**
 - Country: Norway
 - Meter: Kamstrup
 - Encryption enabled: no
 - AMS reader: Pow-K (original ESP8266 version)
 
**Relevant firmware information:**
 - Version: 2.2.12+
 - MQTT: yes
 - MQTT payload type: JSON
 - Temperature sensors: no
 - ENTSO-E API enabled: no

Not really related to the bug, but just wondering:  Is firmware download supposed to be supported?  I lost access to the device after downgrading from 2.2.12 to 2.2.11.  It did not connect to the configured WiFi network, and was also unable to enter AP mode. Even reflashing 2.2.11 over serial did not help. Had to do a flash erase (and reflash) before I got it running 2.2.11. 

Since I have to do every mistake at least twice, I've now tried downgrading from 2.2.13 to 2.2.11.  And lost network again.  No DHCP discovers received.  Haven't yet tried serial since I don't currently have physical access.

Not a big deal, but I guess a small warning in the web GUI would be appropriate if such downgrade issues are to be expected.  If not, then this is probably worth a separate bug report.

ArnieO, 2023-04-13:
Can you check what is the reported reboot reason when running 2.2.12/13?
You can see it on the info page:
[image]

ArnieO, 2023-04-13:
> Not a big deal, but I guess a small warning in the web GUI would be appropriate if such downgrade issues are to be expected. 

After internal discussions, it seems that some downgrades might cause issues. A warning is therefore a good idea. Thank you!

henriklund, 2023-04-13:
I can report a similar behaviour, although I often experience less than 2hrs uptime:
[image].

It has definitely deteriorated with 2.2.12 (anywhere between 11 - 26 daily restarts). However, already with 2.2.11, there was a slight increase compared to previous version  installed 2.2.7)

Last error shown is a Soft WDT reset (3)

henriklund, 2023-04-13:
Just restarted again (1min ago). This time it is a WDT reset (1)

ArnieO, 2023-04-13:
Which hardware are you running, @henriklund - ESP8266 or ESP32?

henriklund, 2023-04-13:
This is ESP8266 (POW-K) in Denmark

ArnieO, 2023-04-13:
> This is ESP8266 (POW-K) in Denmark

OK, thank you! The problem seems to be isolated to ESP8266 then; @gskjold will look into it.

gskjold, 2023-04-15:
Does any of you have debug enabled? This could be a cause for reboots. 

Currently looking into possible reasons, but have yet to find a clue...

gskjold, 2023-04-15:
I have made some changes that have freed a significant amount of heap memory. Please test the firmware below

[esp32.zip](https://github.com/UtilitechAS/amsreader-firmware/files/11238737/esp32.zip)
[esp32s2.zip](https://github.com/UtilitechAS/amsreader-firmware/files/11238738/esp32s2.zip)
[esp32c3.zip](https://github.com/UtilitechAS/amsreader-firmware/files/11238739/esp32c3.zip)
[esp8266.zip](https://github.com/UtilitechAS/amsreader-firmware/files/11238740/esp8266.zip)
[esp32solo.zip](https://github.com/UtilitechAS/amsreader-firmware/files/11238741/esp32solo.zip)

bmork, 2023-04-15:
> Does any of you have debug enabled? This could be a cause for reboots.

I did not.  Only enabled debug after noticing the issue, to see if there were any hints there.  FWIW, enabling debug did not affect the reboot frequency in any noticeable way.

bmork, 2023-04-15:
> Please test the firmware below

Will do as soon as I can, if it's still relevant then. Unfortunately I won't have a chance to test it for couple of weeks.

henriklund, 2023-04-15:
> Does any of you have debug enabled? This could be a cause for reboots.

No.
[image]

henriklund, 2023-04-15:
> I have made some changes that have freed a significant amount of heap memory. Please test the firmware below
> 

Installed. Will post an update within the next few days.

bmork, 2023-04-15:
> Will do as soon as I can, if it's still relevant then. Unfortunately I won't have a chance to test it for couple of weeks.

Found a chance to drive to the meter today anyway.  So I'm also running 02fe207 now

bmork, 2023-04-15:
And I already got a "WDT reset (1)"

henriklund, 2023-04-15:
9 restarts since installation of firmware update. Latest error was a Soft WDT reset (3).

gskjold, 2023-04-15:
Crap.. Wonder what I changed between 2.2.11 and 2.2.12 that made this happen. To be continued I guess

gskjold, 2023-04-15:
By the way, is it always on top of the hour? Check status page for boot timestamp

[image]

henriklund, 2023-04-15:
Only once in the past 12 hrs: (at 17:00). The rest of the time looks completely random (the one just before 12:00 was due to upgrade, and at 12:02 there was another one immediately following the upgrade):
[image]

gskjold, 2023-04-18:
I have deployed a ESP8266 in testing that is constantly running on my meter with MQTT JSON payload, and it does seem to reboot every now and then. To rule out any hardware fault, I hope more of you ESP8266 guys can help run the most recent build to see if it is stable for you.

[esp8266.zip](https://github.com/UtilitechAS/amsreader-firmware/files/11260445/esp8266.zip)

henriklund, 2023-04-18:
Just updated. Will keep you posted.

henriklund, 2023-04-18:
Already had the first reboot - Soft WDT reset (3)

gskjold, 2023-04-19:
I think I finally got to the bottom of this problem, my 8266 unit have been running without issues for 10hrs now. Please try the firmware below:
[esp8266.zip](https://github.com/UtilitechAS/amsreader-firmware/files/11269172/esp8266.zip)

henriklund, 2023-04-19:
No significant change in number of restarts over the past day (since noon on the 18th):
[image]

gskjold, 2023-04-19:
Thanks for the update. Hopefully the firmware I posted  a few minutes ago will show just as much change on your setup as on mine

henriklund, 2023-04-19:
Updated now.
[image]
 Keeping my fingers crossed!

henriklund, 2023-04-19:
Already two reboots since update.

gskjold, 2023-04-19:
Crap... Im assuming you have MQTT enabled? What type of payload?

henriklund, 2023-04-19:
MQTT with JSON + price fetch are enable

henriklund, 2023-04-19:
Just restarted once more (I get a notification on my phone every time this happens). This time I noted a 'Reason' I did not see before: Software restart (4/0)

gskjold, 2023-04-19:
Lets try one more easy change:
[esp8266.zip](https://github.com/UtilitechAS/amsreader-firmware/files/11273434/esp8266.zip)

henriklund, 2023-04-19:
Updated.
[image]
Will keep you posted.

henriklund, 2023-04-20:
So far no reboots. Uptime is the highest in the past 3 week

gskjold, 2023-04-20:
Happy to hear that! My aim is to release this weekend, would be great if you could post an update before tomorrow afternoon.

henriklund, 2023-04-20:
Sure. Will do. So far it keeps on running without any reboots :)

henriklund, 2023-04-20:
> Sure. Will do. So far it keeps on running without any reboots :)

Until approx. 1hrs ago, not a single reboot for the past 24hrs. However, since then the unit is reporting a '(W) No HAN data received last 30s, single blink'.

gskjold, 2023-04-21:
Strange, did you solve this in the end? Sort of sounds like you are not receiving data anymore. I have heard that some Kamstrup users have needed to have their meter rebooted when this happens.

> Even the debug contains more info (although HAN still down)
Yes, I removed this specific debug line, because the output would reveal the username and password if posted in a debug log here.

> Any idea if 'gpioHanPinPullup 1' can affect this HAN behaviour?
This is new in 2.2.13, the default is 1 and will give it the same behavior as in earlier versions. I don't think setting this to 0 could give you any benefit on Pow-K.

henriklund, 2023-04-21:
> Strange, did you solve this in the end? Sort of sounds like you are not receiving data anymore. I have heard that some Kamstrup users have needed to have their meter rebooted when this happens.
> 
No. Actually ended up with a device caught in what appears to be an endless reboot. Not even emergency factory reset helped. I will have to find a FTDI and do a firmware flash.
As for the meter reboot; already posted the question to my provider. Will have to see who is faster (me flashing or them answering).

gskjold, 2023-04-21:
Ouch, sorry to hear that.. There are indeed some conditions with config file upload that can cause reboot loop on previous versions...

ArnieO, 2023-04-21:
For Pow-K users it is fairly easy to see if the meter sends data:
The blue part of the LED is directly connected to the incoming data line (i.e. not controlled by the microcontroller, so indication will be correct even if firmware not running correctly). 
If there is blue blinking each 10 second, the meter delivers data.

henriklund, 2023-04-21:
> For Pow-K users it is fairly easy to see if the meter sends data: The blue part of the LED is directly connected to the incoming data line (i.e. not controlled by the microcontroller, so indication [...]

Then the conclusion would be that the meter does not deliver data (it was just flashing red).

ArnieO, 2023-04-21:
> Then the conclusion would be that the meter does not deliver data (it was just flashing red).

OK, I see - but have difficulties seeing how the old Pow-K with ESP8266 could be the cause of that. 
Anyway, we have seen several cases of Kamstrup meters just stop transmitting (without any changes being done to anything). 

There are two ways of rebooting the meter:
1. Contact grid company, request a remote reboot
2. Cut the main breaker for 30 seconds, and switch back on. This is often a bit awkward, as many household appliances have clocks that will need readjustments.

henriklund, 2023-04-21:
> OK, I see - but have difficulties seeing how the old Pow-K with ESP8266 could be the cause of that.

That was not my perception either, but I had to rule out firmware. Had I known (aka remember) the blue light being indicative of the Pow-K actually receiving data, I would simply have waited for my meter company.
Nevertheless, the oddity still remains that after me triggering a restart (on the last beta software above), the device did not appear to perform NTP operation (despite trying a local NTP server as well). 'Updating' to 2.2.13 resolved that matter. That being said, the version was (from a reboot PoV) significantly more stable that the past versions.

henriklund, 2023-04-21:
> Cut the main breaker for 30 seconds, and switch back on. This is often a bit awkward, as many household appliances have clocks that will need readjustments

That is unfortunately not possible since the meter is directly coupled on the main line before the breaker. Flipping the main breaker will thus have no effect.

ArnieO, 2023-04-21:
> That is unfortunately not possible since the meter is directly coupled on the main line before the breaker. Flipping the main breaker will thus have no effect.

Ah, OK. Apparently not same setup as we have in Norway then. One customer in Norway contacted the grid company and requested a reset of his Kamstrup. They declined and asked him to to it himself - using the main fuse.

PS: 
I never heard of any other meter brand than Kamstrup that need reboots from time to time. Apparently a weakness in its HW or firmware.

closed 2023-04-21 by gskjold