UtilitechAS/amsreader-firmware #524: JSON error results in "Loading configuration" forever.

kasperlaursen, 2023-04-12:
**Describe the bug**
Hi 👋 
Thanks for making the great tool! 

I just flashed `ams2mqtt-esp8266-2.2.13` on a esp8266 (NodeMcu v3 by Lolin).
I do get the current Wattage in the dashboard, but when opening the configuration, i am stuck on the "Loading configuration" screen. 

In the chrome console i get the following error: `VM16:1 Uncaught (in promise) SyntaxError: Bad control character in string literal in JSON at position 1119`.

Attached is the log + the VM16.json that gets the error. 

**To Reproduce**
Steps to reproduce the behavior:
1. Flash the `ams2mqtt-esp8266-2.2.13.bin` downloaded from https://github.com/UtilitechAS/amsreader-firmware/releases
2. Setup the Pinout and board
3. Connect to local SSID
4. Go to Configuration page

**Expected behavior**
The configuration is loaded, and the fields are filled. 

**Screenshots**
[image]

**Hardware information:**
 - Country: Denmark
 - Meter: Kamstrup - 6841138BN143101011
 - Encryption ? (I didn't enter it, but config does not load, so i do not know)
 - AMS reader: Generic ESP8266
 - M-bus adapter (if applicable):

**Relevant firmware information:**
 - Version: v2.2.13
 - MQTT: defaults
 - MQTT payload type: default
 - HAN GPIO: GPIO15 (D5)
 - HAN baud and parity: default
 - Temperature sensors default
 - ENTSO-E API enabled: default

**Additional context**
[elmaler.local-1681313510342.log](https://github.com/UtilitechAS/amsreader-firmware/files/11213200/elmaler.local-1681313510342.log)
[VM16.json.zip](https://github.com/UtilitechAS/amsreader-firmware/files/11213214/VM16.json.zip)

kasperlaursen, 2023-04-12:
> Same problem as this? [#519 (reply in thread)](https://github.com/UtilitechAS/amsreader-firmware/discussions/519#discussioncomment-5594398)

Seems so yes!
Didn't think to check the discussion section before posting, only searched the issues 🤦‍♂️

kasperlaursen, 2023-04-12:
> For anyone with this same issue. I intercepted the Fetch of `configuration.json` in the network tab. Then i removed the broken characters in the bottom of the json object, and returned the rest.
> 
> I guessed the broken values, and saved the config. Now the config loads correctly in subsequent loads.
> 
> For developers, I re-flashed the ESP a few times to see if that would fix it, but the samme issue happened multiple times. Just to let you know it was not a fluke :)

After a reload it seems the issue (broken characters in the config) is back.
However my saved values are there as well and seems to be saved successfully.

ArnieO, 2023-04-12:
Sounds like a bug that @gskjold will surely fix as soon as he finds time.

gskjold, 2023-04-15:
Caused by not clearing Home Assistant config by default

closed 2023-04-15 by gskjold