UtilitechAS/amsreader-firmware #525: PriceAPI: Exceeded API rate limit

StarkJohan, 2023-04-15:
Yesterday (14/4 after nordpool prices was released) the API was unresponsive, can't remember the exact error message but I assumed it was temporary, it was something in the lines of "API not responding". This morning the message has changed and might be more problematic perhaps... I assume I'm not the only one with this issue but I thought it would be best to report it just in case.

**PriceAPI: Exceeded API rate limit**

gskjold, 2023-04-15:
We use ENTSO-E as our source for prices, and they have not released the prices for 15.apr which in turn makes all the devices poll for new prices at a higher rate than our server is currently configured to allow, this causes the client to show this error. Adjustments will be made on both firmware and server to deal with similar situations.

Please keep this issue open until the problem is resolved

StarkJohan, 2023-04-15:
I figured that might be the case. When making changes to account for this situation please note that this also affects the HA MQTT sensors which report empty values.

`Invalid state message '{"id":"C4:5B:BE:XX:XX:XX","prices":{"0":-127.0000,"1":-127.0000,"2":-127.0000,"3":-127.0000,"4":-127.0000,"5":-127.0000,"6":-127.0000,"7":-127.0000,"8":-127.0000,"9":-127.0000,"10":-127.0000,"11":-127.0000,"12":-127.0000,"13":-127.0000,"14":-127.0000,"15":-127.0000,"16":-127.0000,"17":-127.0000,"18":-127.0000,"19":-127.0000,"20":-127.0000,"21":-127.0000,"22":-127.0000,"23":-127.0000,"24":-127.0000,"25":-127.0000,"26":-127.0000,"27":-127.0000,"28":-127.0000,"29":-127.0000,"30":-127.0000,"31":-127.0000,"32":-127.0000,"33":-127.0000,"34":-127.0000,"35":-127.0000,"36":-127.0000,"37":-127.0000,"min":0.0000,"max":0.0000,"cheapest1hr":"","cheapest3hr":"","cheapest6hr":""}}' from 'homeassistant/kamstrup/prices'`

gskjold, 2023-04-15:
Prices are now available, and better handling in code have been added

closed 2023-04-15 by gskjold