UtilitechAS/amsreader-firmware #527: Previous month accumalated values

tronde-ams, 2023-04-16:
Is it possible to make the accumulated values for energy and total price for the previous month available in some way?

As of now, the accumulated values disappear when a new month begin, and it would have been nice to have them available for some time after.

I know it will add clutter to the GUI, but can it be made available via URL (as most values already are) or by MQTT?

ArnieO, 2023-04-17:
I think that is an interesting proposition, and it would not necessarily clutter the GUI (much).
It could be done as a new graph on the bottom, maybe as default turned off. A rolling last 12 months maybe? Or maybe just keep accumulating as the months roll on? It could maybe be a double-bar graph with one vertical axes indicating consumption, the other indicating estimated cost (total price)? 

When we started making the data available graphically, we were mainly focused on the short term, not so much on the longer term, as that data is usually made available by the electricity company - or handled by user's smart home system. 

We will take this request into consideration, in the mean time it could be nice to hear what the user community thinks.

tronde-ams, 2023-04-17:
I would be more than happy with something like this.

.
[image]

gskjold, 2023-04-21:
This is a good idea. I actually have the last month cost stored, but I see that I am missing last month kwh, so I will need to add that. I'll try to get that done in a future patch, too much to add for this weekends release.

closed 2023-05-04 by gskjold