UtilitechAS/amsreader-firmware #528: Wrong accumulated values of consumption

spinehead, 2023-04-18:
**Describe the bug**
Hour consumption value seem ok, however accumulated day and month is not shown correctly. 
Same for cost, day and month (hour is ok)
Tariff peaks and the graphs daily consumtion and monthly consumtion is not shown/only sporadic showing values.
There are current transfomers on the meter, so multiplier of 100 is used for watt and amps (100 is confirmed by supplier)
Meter stand shown under import graf is same as on the Aidon meter.

**To Reproduce**
Steps to reproduce the behavior:
Nothing special is requied to reproduce the issue. It's constant

**Screenshots**
If applicable, add screenshots to help explain your problem.
[image]
[image]
[image]

**Hardware information:**
 - Country: Norway
 - Meter: Aidon
 - Encryption enabled no
 - AMS reader: Pow-U
 - M-bus adapter (if applicable):

**Relevant firmware information:**
 - Version: 2.2.13.
 - MQTT: no
 - MQTT payload type: NA
 - HAN GPIO: [e.g. GPIO5]
 - HAN baud and parity: 2400 8E1
 - Temperature sensors NA
 - ENTSO-E API enabled: yes

**Additional context**
Add any other context about the problem here.

ArnieO, 2023-04-18:
We have seen some such cases before. Our current hypothesis is that the root cause could be EEPROM read/write errors.

We're considering to remedy this by implementing a duplicate/redundant write of stored data to EEPROM, combined with CRC check (control sum). If CRC check fails, the duplicate/redundant data will be used. This is work in progress, not implemented yet.

You can edit the stored data by following the procedure indicated here: https://amsleser.no/blog/post/28-correcting-stored-values-by-editing-the-configuration-file

gskjold, 2023-04-18:
When did you upgrade to 2.2.13 ? I'm just wondering if it is days or hours ago

spinehead, 2023-04-18:
I upgraded 5 hours ago. The issue was there with previous versions as well. 
I got a beta version that fixed partially the issue, see #448

I noticed now that the cost (last day) seem not to accumulate, but is somehow reset to a sub zero value after turining into a new hour. Screenshot is taken approx. 13:05

[image]

gskjold, 2023-04-18:
OK, I understand. There was an issue in 2.2.11 and earlier that could make the values on the graphs go crazy, this have been resolved in 2.2.12, but it does not fix previously corrupted data, so you either have to modify the existing data, factory reset or wait for the incorrect data to disappear.

The cost for day is calculated based on the data in the graphs, so my assumption is that the day cost rolls around because of that large spike at midnight.

spinehead, 2023-04-19:
I edited the data and uploaded to the reader. However the reader became none accessible afterwards. Tried both reset and flashed new firmware still same. The indicating lights seem to work as normal, however I can't access the reader by wifi anymore. Light sequences when perfoming reset / activate AP is as described in manual. Any good advice? :)

gskjold, 2023-04-19:
Have you tried a emergency factory reset? When light is green during boot, click and hold AP button, it will then turn yellow and then red. Release button when red and it will confirm factory reset with red blinks. AP should be active after that

spinehead, 2023-04-20:
I'm not able to emergency factory reset it.. The sequence (multicolour LED besides the bus port) of lights is: when powered , blue is always on, then green stable light is on for 1-2 s and then red light has to rapid blinks before both green and red is off. This repeats continousely. When I try to do EFR, blue light is on all time when pushing reset buttom. The EFR seems to be exceuted as the lights (green on and then push AP and release it when led turn red, gives rapid short sequence of red blinks. However, still not able to connect to the SSID. The blue LED is always ON through the EFR sequence btw.  This is an "old" reader, so I have ordered a new one today :) Thanks for quick feedback

ArnieO, 2023-04-20:
OK, my understanding is then that your oldest Pow-U was bought in November -21 - so it is a device with ESP8266. We're having some issues with ESP8266 devices on the latest firmware versions, causing a reboot loop, there is a test firmware in this thread: https://github.com/UtilitechAS/amsreader-firmware/issues/523

Unfortunately, you will need an FTDI module to reflash the old ESP8266-based Pow-U devices (see [User Manual](https://amsleser.no/index.php?controller=attachment&id_attachment=47) paragraph 3.2.2).

spinehead, 2023-04-21:
Thanks, It worked!

ArnieO, 2023-04-21:
> Thanks, It worked!

Good to hear! We're working towards an official release implementing the fix this weekend.

closed 2023-04-21 by gskjold