UtilitechAS/amsreader-firmware #531: Support for small OLED display like SSD1306

toluen-kodekrets, 2023-04-20:
What about including support for a small display like SSD1306. This would be usefull for elderly people/ or people not wanting to use app/browser, but still getting insight in to their power usage. The display could show their current usage live.

[image]

Alilink:
[https://www.aliexpress.com/item/32638662748.html?spm=a2g0o.order_list.order_list_main.4.56a81802YHAQ5v](url)

kaihp, 2024-01-29:
The SSD1306 is not a display, but a display controller. I have code for initialization and how to display pixels, that I can donate.

alexwala, 2024-07-08:
https://github.com/UtilitechAS/amsreader-firmware/issues/806#issuecomment-2210825149

display code (using TFT_eSPI / lvgl) already included to show ActiveImportPower / ActiveImportCounter :-)

alexwala, 2024-07-12:
tried to adapt the code for a [LILYGO T-Dongle S3](https://www.lilygo.cc/products/t-dongle-s3):

- onboard RGB Led using [FastLED](https://github.com/FastLED/FastLED)
- display using [TFT_eSPI](https://github.com/Bodmer/TFT_eSPI), [lvgl](https://lvgl.io/)

[image]