UtilitechAS/amsreader-firmware #535: Failed to get future price. PriceAPI: Unspecified error -1

viktor08, 2023-04-22:
After upgrading to 2.2.14, the POWK+ fails to get the future price form the API.
I've tried to generate a new API-key at ENTOS-E, but the same error persist.

[image]

**Hardware information:**
 - Country: Norway
 - Meter: Kamstrup
 - Encryption enabled: no
 - AMS reader: PowK+, esp32s2

**Relevant firmware information:**
 - Version: 2.2.14
 - MQTT: yes
 - MQTT payload type: HomeAssistant
 - HAN baud and parity: 2400 8N1
 - ENTSO-E API enabled: yes/no

gskjold, 2023-04-23:
I have a test unit running with entso-e token already, but I am unable to reproduce your issue.

Apparently, -1 is "Connection refused" (https://github.com/espressif/arduino-esp32/blob/237a3fe96c451e5da88e8cfe9d850961f9bbe036/libraries/HTTPClient/src/HTTPClient.h#L45)
There is a number of network related issues that could cause this, like firewall etc.

Did this get fixed for you? If you run debugging with "Debug" level, the URL will be visible in the debugger, and you can run it manually in your browser to check there. Keep in mind that the URL contains your token, so don't post it here. (https://amsleser.no/blog/post/24-telnet-debug)

ArnieO, 2023-04-23:
@gskjold 
We have a customer reporting same error code by email a few days ago (before 2.1.14), so don't think this is related to the update.

IRQNMI, 2023-04-23:
Same issue after upgrading to 2.2.14 - I think its related with time issue - system time of by one hour after upgrade

[image]
(Snippet taken at 12:11)

ArnieO, 2023-04-23:
> Same issue after upgrading to 2.2.14 - I think its related with time issue - system time of by one hour after upgrade
> 
> [image] (Snippet taken at 12:11)

What is your time zone setting?

IRQNMI, 2023-04-23:
[image]

Running: "w32tm /stripchart /computer:pool.ntp.org" from a computer on the same nettwork produces:

Tracking pool.ntp.org [162.159.200.123:123].
The current time is 23.04.2023 12:28:17.
12:28:17, d:+00.0259496s o:-00.1654725s  [                           *                           ]
12:28:19, d:+00.0263884s o:-00.1659268s  [                           *                           ]
12:28:21, d:+00.0263088s o:-00.1649046s  [                           *                           ]

So there should  not be any connection issues with the ntp server(s)

ArnieO, 2023-04-23:
The customer I referred to in https://github.com/UtilitechAS/amsreader-firmware/issues/535#issuecomment-1519001567 reports that his issue was solved by changing DNS to 1.1.1.1. - at least when not using Entso-E API key. 
Could you try that?

IRQNMI, 2023-04-23:
- Changed gw router from using dchp/isp dns to dns servers 1.1.1.1 (Cloudfare)
- Rebooted gw router
- Rested dns working
- Rebooted AMS reader
- waited for full hour (14:00)
  (during wait time - the date/time in GUI are showing 01.01.1970 01:xx - where xx are are same as uptime since reboot)

After coming to full hour - 14:00 - I get the "PriceAPI: Unspecified error -1" and the date/time are showing, with red font, 23.04.2023 13:00

IRQNMI, 2023-04-23:
- Changed AMS Reader setting from DHCP to Static IP with DNS server 1.1.1.1. Clock is still 1 hour off - but I get price information!
- With Static IP still set - i changed DNS from 1.1.1.1 to my gw - same result (getting price info and clock is 1 hour off)
- Changed internet route / GW back to ISP DHCP/DNS settings (130.67.15.198/193.213.112.4 (Telenor)) - Missing price information and clock is back to showing uptime

No clear one failure - multiple failures? - both in DNS and NTP handling?

(Price comes from multiple sources? dicretly from ENTSO-E (I have API key) and from your proxy (time off should give SSL trouble with ENTSO)?)

IRQNMI, 2023-04-23:
I have tried to reproduce what I did - and I'm unable to do so..
#536 - points to an error in memory handling / mapping... 
So the problem is dependent on which data/config are stored where in memory

ArnieO, 2023-04-23:
Yeah, there seems to be something here - triggered by... something. I'm sure @gskjold will look into it when he finds time. 
Fortunately most of us do not see any issues.

ArnieO, 2023-04-23:
I just realized I have the same issue:
[image]

It seems like it is only the timer (red font) that is wrong, everything else looks normal. 
Never seen this before. 

I use static IP and external DNS.
[image]

henriklund, 2023-04-23:
@viktor08, there is an issue with odd DNS requests (#536). Are you able to run WireShark and check whether DNS lookup sent from device are valid?

gskjold, 2023-04-24:
This is very good information, thanks for being thorough! Since the time gets set at the top of the hour, it suggests that it received the time from the meter and not NTP. I know we have an issue with Kamstrup sending CET timestamp from their meter all year, so during CEST the time will be 1hr off.

So the conclusion is that there is an issue with both NTP and DNS under certain conditions. Maybe they are connected as well. By default the NTP settings have a hostname set, so using an IP address may fix the NTP issue (ex: 129.240.2.6)

I will set up some testing locally one day I have time and dive into this issue.

ArnieO, 2023-04-24:
I have the same error whether I use IP or hostname for the device:
[image]
[image]

I also tried the IP you suggested for NTP, but no change after having passed top of hour:
[image]

(I passed top of hour with IP on both device and NTP)

PerBuch54, 2023-04-24:
Same issue here in DK Pow-P1 on a l+g meter.
Came after a reboot 
Last boot: 24.04.2023 03:01
Reason: Vbat power on reset (1/0)
PriceAPI: Server says its too early.
Time one hour ahead.

nossaile, 2023-04-24:
I discovered the same problem this morning with my Pow-P1 (v2.2.14) connected to L&G E360: Time one hour ahead (red numbers), no values from ENTSO-E and some historic values zeroed. Reboot didn’t help. 

Downloaded and installed v2.2.13 and the time and ENTSO-E values are correct displayed.

PerBuch54, 2023-04-24:
Solves the issue here :-)

IRQNMI, 2023-04-24:
I'm also confirming that version: 6e8233e - works fine with both NTP/Time issue and ENTSO/price issue :)
Chip: esp32s2
Device: Pow-K+

IRQNMI, 2023-04-24:
I see the commit are only doing changes on the handling of time from the meter - where, if any places, are the NTP used? PCAP in the other case, showed strange requests?

gskjold, 2023-04-24:
A previous commit https://github.com/UtilitechAS/amsreader-firmware/commit/ed9cf4b87d17756fe2bf9e641e2d98896aa4d93b fixes the NTP issue. 

Using a static field for the server hostname is the fix
https://github.com/UtilitechAS/amsreader-firmware/blob/6e8233e4af83a65effd9767ce0d295a7670aec04/src/AmsToMqttBridge.ino#L560

Fix is based on information found here: https://github.com/espressif/arduino-esp32/issues/6720

closed 2023-04-25 by gskjold