UtilitechAS/amsreader-firmware #536: No time/date information after 2.2.14 upgrade

tubbelino, 2023-04-23:
**Describe the bug**
After 2.2.14 upgrade NTP time/date stopped working

**To Reproduce**
Install or upgrade to version 2.2.14 on Wemos D1 mini.

**Expected behavior**
Time and date is not updated in AMSreader

**Screenshots**
[image]

[image]

**Hardware information:**
 Tested two types of Wemos D1 mini, one original and one clone.

**Relevant firmware information:**
 - Version: 2.2.14

**Additional context**
Packet sniffing shows strange DNS query output from the AMS reader, pcap file is available.
Tested with both DHCP and Static IP and different ntp servers.

henriklund, 2023-04-23:
I had a similar experience (ie. clock not being updated) after a downgrade from a beta version (trying to troubleshoot frequent reboots) to v2.2.13. I did not think of running a WireShark to troubleshoot. Since I managed to brick the device within the same hour I had to resort to reflash the device (ESP8266). After the reflash, everything was OK.

ArnieO, 2023-04-24:
Interesting comment @henriklund: I also went from a beta version to 2.2.14, and now have the clock error described in this thread: https://github.com/UtilitechAS/amsreader-firmware/issues/535.

ArnieO, 2023-04-24:
I took my Pow-K out of the meter and did a factory reset - and it now does not display date/time at all.
So it looks like the date/time in the top bar is no longer from internet, but from the meter. 
[image]

gskjold, 2023-04-24:
Are all of you on ESP8266?

gskjold, 2023-04-24:
After rebooting my devices, this also happens here...

gskjold, 2023-04-24:
Test version below:

[esp32solo.zip](https://github.com/UtilitechAS/amsreader-firmware/files/11311113/esp32solo.zip)
[esp32.zip](https://github.com/UtilitechAS/amsreader-firmware/files/11311114/esp32.zip)
[esp8266.zip](https://github.com/UtilitechAS/amsreader-firmware/files/11311115/esp8266.zip)
[esp32c3.zip](https://github.com/UtilitechAS/amsreader-firmware/files/11311116/esp32c3.zip)
[esp32s2.zip](https://github.com/UtilitechAS/amsreader-firmware/files/11311117/esp32s2.zip)

ThorAlex87, 2023-04-24:
I'll try the test version, mine has been 15 minutes late since I updated yesterday.

tubbelino, 2023-04-24:
Test version looking good so far for me!

Wemos D1 mini (ESP8266EX)

closed 2023-04-25 by gskjold