UtilitechAS/amsreader-firmware #539: Amperage scaling for Kaifa MA309MH4LAT1 Austria/Vorarlbergnetz

Rasfumi, 2023-04-25:
Hi, first i want to thank you for this great project!

I have the firmware running since last November on a ESP8266 connected to my Smart meter: Kaifa MA309MH4LAT1 in Austria supplied by provider "Vorarlbergnetz".

I just updated to V2.2.15 from an unknown version of Nov 2022. Unfortunately the amperage scaling is not correct anymore now. The used multipliers are: Watt: 1, Volt: 10, Amp: 100, kWh: 1.

The amperes displayed are too low by a factor of 10 i think but it is not possible to set the scaling to 1000 as it is limited to 655.
Would it be possible to set the limit higher in the webpage?

Thank you

Rasfumi, 2023-04-25:
This is what i get when i dest telnet debug to verbose. After this data the ESP8266 reboots.

* Debug: Command received: v
* Debug level set to Verbose
(V) MBUS frame:
(V) 68 FA FA 68  53 FF 00 01  67 DB 08 4B  46 4D 10 20
(V) 02 C8 F5 82  01 55 21 00  2A A7 66 A2  92 70 DA 48
(V) 91 B7 6C 58  DC DB 73 37  C7 E1 80 99  61 88 29 07
(V) FD D5 0A 44  6D 39 00 A5  FC 51 D9 1E  6F 10 4E 55
(V) 43 34 78 90  F3 10 DC 72  E9 3A 53 EE  41 9F EF C6
(V) E7 A6 58 C2  47 F0 7C 5D  47 5B B4 27  DD 22 6E FB
(V) 82 A5 50 E5  C7 2A D7 16  45 8E 6A 1A  75 02 8F E3
(V) 44 B7 F5 C2  07 70 69 9E  09 B5 7D A4  FB 81 B3 E2
(V) 8A 7B 65 E0  27 2F 87 69  A6 88 7B 63  8B 7B A8 0F
(V) D6 9C 45 4F  00 26 1E 55  74 BA 97 FE  8B CD F8 A0
(V) FB 23 4E AA  68 40 7B 65  DF 7B DA 8D  3B 73 DF AD
(V) C9 DD B9 44  A0 1C 19 46  6A 43 D0 C8  47 66 5B 2E
(V) D2 51 C4 6A  B4 CE 27 FD  31 D6 13 F3  CB CC F6 98
(V) 00 CC 19 9B  1F FF 75 F6  38 1A F3 28  F8 15 6F 94
(V) 3A 6C E1 62  7A A2 75 B8  59 B7 90 EE  02 EC 41 DB
(V) 35 E2 36 4C  F8 D6 19 FF  C3 55 79 C6  78 AB 66 16
(V)
(V) MBUS frame:
(V) 68 72 72 68  53 FF 11 01  67 B0 DB 4A  0A 20 70 45
(V) 53 A4 9F 68  62 4E D5 9E  3D 5C 8D 4E  6A 58 9B 45
(V) D0 74 FC 98  D5 34 EC E2  8E 3C 7D CD  92 FB 2D 5B
(V) AA 05 00 B3  60 A1 28 9F  9D 8B 5C EF  68 19 53 EF
(V) 65 ED 9C B2  83 AA 32 E8  7D CE DA DB  82 8A F0 80
(V) 53 B9 E1 15  AA 36 BD 75  6A 67 7A 1F  2B A5 66 2B
(V) D9 1B 93 62  A8 29 91 32  14 52 C7 58  05 2F 7B DD
(V) 10 26 CD 36  95 BD 49 16
(V) GCM frame:
(V) DB 08 4B 46  4D 10 20 02  C8 F5 82 01  55 21 00 2A
(V) A7 66 0F 00  2A 3C B4 0C  07 E7 04 19  02 0A 3B 1E
(V) 00 FF 88 80  02 10 09 06  00 00 01 00  00 FF 09 0C
(V) 07 E7 04 19  02 0A 3B 1E  00 FF 88 80  02 02 09 06
(V) 00 00 60 01  00 FF 09 0E  31 4B 46 4D  30 32 30 30
(V) 31 38 32 35  31 37 02 02  09 06 00 00  2A 00 00 FF
(V) 09 10 4B 46  4D 31 32 30  30 32 30 30  31 38 32 35
(V) 31 37 02 03  09 06 01 00  20 07 00 FF  12 09 1A 02
(V) 02 0F FF 16  23 02 03 09  06 01 00 34  07 00 FF 12
(V) 09 15 02 02  0F FF 16 23  02 03 09 06  01 00 48 07
(V) 00 FF 12 09  1A 02 02 0F  FF 16 23 02  03 09 06 01
(V) 00 1F 07 00  FF 12 00 2D  02 02 0F FE  16 21 02 03
(V) 09 06 01 00  33 07 00 FF  12 00 2B 02  02 0F FE 16
(V) 21 02 03 09  06 01 00 47  07 00 FF 12  00 51 02 02
(V) 0F FE 16 21  02 03 09 06  01 00 01 07  00 FF 06 00
(V) 00 00 00 02  02 0F 00 16  1B 02 03 09  06 01 00 02
(V) 07 00 FF 06  00 00 00 18  02 02 0F 00  16 1B 02 03
(V) 09 06 01 00  01 08 00 FF  06 00 15 68  9C 02 02 0F
(V) 00 16 1E 02  03 09 06 01  00 02 08 00  FF 06 00 10
(V) D5 46 02 02  0F 00 16 1E  02 03 09 06  01 00 03 08
(V) 00 FF 06 00  02 04 D1 02  02 0F 00 16  20 02 03 09
(V) 06 01 00 04  08 00 FF 06  00 07 D3 2A  02 02 0F 00
(V) 16 20

(V) DLMS frame:
(V) 0F 00 2A 3C  B4 0C 07 E7  04 19 02 0A  3B 1E 00 FF
(V) 88 80 02 10  09 06 00 00  01 00 00 FF  09 0C 07 E7
(V) 04 19 02 0A  3B 1E 00 FF  88 80 02 02  09 06 00 00
(V) 60 01 00 FF  09 0E 31 4B  46 4D 30 32  30 30 31 38
(V) 32 35 31 37  02 02 09 06  00 00 2A 00  00 FF 09 10
(V) 4B 46 4D 31  32 30 30 32  30 30 31 38  32 35 31 37
(V) 02 03 09 06  01 00 20 07  00 FF 12 09  1A 02 02 0F
(V) FF 16 23 02  03 09 06 01  00 34 07 00  FF 12 09 15
(V) 02 02 0F FF  16 23 02 03  09 06 01 00  48 07 00 FF
(V) 12 09 1A 02  02 0F FF 16  23 02 03 09  06 01 00 1F
(V) 07 00 FF 12  00 2D 02 02  0F FE 16 21  02 03 09 06
(V) 01 00 33 07  00 FF 12 00  2B 02 02 0F  FE 16 21 02
(V) 03 09 06 01  00 47 07 00  FF 12 00 51  02 02 0F FE
(V) 16 21 02 03  09 06 01 00  01 07 00 FF  06 00 00 00
(V) 00 02 02 0F  00 16 1B 02  03 09 06 01  00 02 07 00
(V) FF 06 00 00  00 18 02 02  0F 00 16 1B  02 03 09 06
(V) 01 00 01 08  00 FF 06 00  15 68 9C 02  02 0F 00 16
(V) 1E 02 03 09  06 01 00 02  08 00 FF 06  00 10 D5 46
(V) 02 02 0F 00  16 1E 02 03  09 06 01 00  03 08 00 FF
(V) 06 00 02 04  D1 02 02 0F  00 16 20 02  03 09 06 01
(V) 00 04 08 00  FF 06 00 07  D3 2A 02 02  0F 00 16 20
(V)

(D) Received valid DLMS at 36

(V) Using application data:
(V) 02 10 09 06  00 00 01 00  00 FF 09 0C  07 E7 04 19
(V) 02 0A 3B 1E  00 FF 88 80  02 02 09 06  00 00 60 01
(V) 00 FF 09 0E  31 4B 46 4D  30 32 30 30  31 38 32 35
(V) 31 37 02 02  09 06 00 00  2A 00 00 FF  09 10 4B 46
(V) 4D 31 32 30  30 32 30 30  31 38 32 35  31 37 02 03
(V) 09 06 01 00  20 07 00 FF  12 09 1A 02  02 0F FF 16
(V) 23 02 03 09  06 01 00 34  07 00 FF 12  09 15 02 02
(V) 0F FF 16 23  02 03 09 06  01 00 48 07  00 FF 12 09
(V) 1A 02 02 0F  FF 16 23 02  03 09 06 01  00 1F 07 00
(V) FF 12 00 2D  02 02 0F FE  16 21 02 03  09 06 01 00
(V) 33 07 00 FF  12 00 2B 02  02 0F FE 16  21 02 03 09
(V) 06 01 00 47  07 00 FF 12  00 51 02 02  0F FE 16 21
(V) 02 03 09 06  01 00 01 07  00 FF 06 00  00 00 00 02
(V) 02 0F 00 16  1B 02 03 09  06 01 00 02  07 00 FF 06
(V) 00 00 00 18  02 02 0F 00  16 1B 02 03  09 06 01 00
(V) 01 08 00 FF  06 00 15 68  9C 02 02 0F  00 16 1E 02
(V) 03 09 06 01  00 02 08 00  FF 06 00 10  D5 46 02 02
(V) 0F 00 16 1E  02 03 09 06  01 00 03 08  00 FF 06 00
(V) 02 04 D1 02  02 0F 00 16  20 02 03 09  06 01 00 04

dbeinder, 2023-04-25:
Vorarlbergnetz encodes a scaling factor with each value, I have implemented decoding some time ago here: https://github.com/dbeinder/KsmWestKaifaReader/blob/master/kaifa.py#L175-L222

gskjold, 2023-04-25:
The same scaling factor is handled in our code here: https://github.com/UtilitechAS/amsreader-firmware/blob/master/src/IEC6205675.cpp#L590-L593

So, I'm a bit puzzled why you get the wrong numbers here, you should not need any scaling factor on anything really

gskjold, 2023-04-25:
Correction, you should not need any multiplier in config

gskjold, 2023-04-25:
Ah, scratch that, I found the problem! Will build you a test firmware in a sec

gskjold, 2023-04-25:
[esp32.zip](https://github.com/UtilitechAS/amsreader-firmware/files/11321215/esp32.zip)
[esp32c3.zip](https://github.com/UtilitechAS/amsreader-firmware/files/11321216/esp32c3.zip)
[esp32s2.zip](https://github.com/UtilitechAS/amsreader-firmware/files/11321217/esp32s2.zip)
[esp32solo.zip](https://github.com/UtilitechAS/amsreader-firmware/files/11321218/esp32solo.zip)
[esp8266.zip](https://github.com/UtilitechAS/amsreader-firmware/files/11321220/esp8266.zip)

Rasfumi, 2023-04-25:
Wow, that was fast! Thanks a lot for fixing!
[image]

gskjold, 2023-04-25:
No problem :) Will include this in 2.2.16. It was a old special case handling added for Norwegian Kaifa meters. This handling have been changed in later versions, so the code have technically been dormant until recently.

closed 2023-04-25 by gskjold