UtilitechAS/amsreader-firmware #540: Dark mode

gskjold, 2023-04-25:
There have been quite a few questions about this. Currently being implemented by one of our team members. I suspect a test version will be posted in the near future.

ElMyggo, 2023-04-25:
Nice :-) 
(I have my own build with dark mode but its not perfect... and have to change everything if i want to install a new version...)

gskjold, 2023-04-28:
Pull request merged. Modified to store the configuration on flash. Build attached. We are happy to get some opinions on color palette. This change will be part of the ongoing development with v2.3

[esp32.zip](https://github.com/UtilitechAS/amsreader-firmware/files/11355956/esp32.zip)
[esp32c3.zip](https://github.com/UtilitechAS/amsreader-firmware/files/11355957/esp32c3.zip)
[esp32s2.zip](https://github.com/UtilitechAS/amsreader-firmware/files/11355959/esp32s2.zip)
[esp32solo.zip](https://github.com/UtilitechAS/amsreader-firmware/files/11355960/esp32solo.zip)
[esp8266.zip](https://github.com/UtilitechAS/amsreader-firmware/files/11355961/esp8266.zip)

ElMyggo, 2023-04-28:
I build my own file from dev 2.3 files and flash direct from Visual studio to my test esp32 and i cant save the darkmode status, everytime i open config its disabled.. must i use your build?
[image]

gskjold, 2023-04-28:
Should definitely work building your own, but I can see that you dont have the latest commit, please update your branch and build again

ArnieO, 2023-04-30:
Looks great! 👍

ElMyggo, 2023-05-03:
Love the dark mode! 
but i like the combo black/orange so i use photoshop to test!
(text and number should be black when orange is the background otherwise white)
Possible to change color in config would be a blast!!

[image]

ArnieO, 2023-05-03:
> Possible to change color in config would be a blast!!

I disagree, as the config page is already quite crowded.

ElMyggo, 2023-05-04:
> I disagree, as the config page is already quite crowded.

Yes that's true

spacecosmos, 2023-09-30:
Not a solution, but merely an alternative option. I use the Opera web browser on my phone and it has an option to force dark mode on all pages/sites I'm browsing. And it works (and looks) beautifully! :)

A few examples:

[image]

[image]

[image]

Maybe an option to toggle dark mode on/off on the bar would be nice (at some point):

[image]

Keep up the good work! :)

closed 2023-12-25 by gskjold