UtilitechAS/amsreader-firmware #541: Switching to AP does not work when connected

dbeinder, 2023-04-25:
**Describe the bug**
AP mode works when booting for the first time, but when Wifi is configured, switching to AP using the button fails to create an access point.

Even though the `WiFi.softAP()` and `WiFi.mode()` return `true`, the access point simply doesn't show up.

It seems to be affected by timing, activating debugging or adding a random debug prints can make the problem go away.
For me, switching becomes 100% reliable after adding a 10ms sleep in `swapWifiMode()`: https://github.com/dbeinder/AmsToMqttBridge/commit/a714d88ca2a5a4e24fb099dea45fadf63424ed3d

I didn't find anything in the docs that this is required, so this kind of fix doesn't feel quite right to me 😕

**Hardware information:**
 - Meter: Kaifa MA309
 - AMS reader: µHAN mosquito / ESP32-C3-MINI-1

**Relevant firmware information:**
 - Version: 8b97544a2c7d4ecbaf3d4fa7f38b9090afe68250

gskjold, 2023-04-26:
Thanks, I'll do some testing on this!

> I didn't find anything in the docs that this is required, so this kind of fix doesn't feel quite right to me

Well, same here for auto modem sleep: https://github.com/UtilitechAS/amsreader-firmware/blob/8b97544a2c7d4ecbaf3d4fa7f38b9090afe68250/src/AmsToMqttBridge.ino#L534

There are several examples online where this is stated as "needed", but no official sources to confirm why, but I believe it has to do with yielding CPU time to internal processes to complete certain functions. In some cases you end up in a hardware WDT reboot if you do not have enough delay in your loop to allow subsystems to complete their tasks.

Also, I found a reference somewhere that you need to add delay after analogRead() as well to get the correct reading. And that the amount of delay was dependent on the resistor bridge on the input.

I wish I understood more of this sorcery, but for now I just have to accept that it works and focus on other tasks.

ArnieO, 2023-04-26:
I have never seen this, and have regularly gone to AP on devices that already have wifi configured. Must have done it many tens of times on various firmware rev's over the last 2 years. Have not yet tried it yet on v 2.2.15, but it worked as it should as late as 2.2.14.

dbeinder, 2023-04-27:
First I thought it just a quirk of the ESP32-C3, but it seems to be more than that.
I tested my small batch and there is some variation, ~90% switch AP/STA without problem, the rest consistently show this issue. I'll have to make some measurements and see if this is variation in my board or the ESP itself.

ArnieO, 2023-04-27:
Can you provide some more description of your hardware? 
I suspect the issue could be related to powering of the ESP. The ESP consumes significantly more power when AP is active, and your design might just be marginally good enough (since some units work). 

It is me that has designed the hardware of Pow-K, Pow-U and Pow-P1 that we sell on https://amsleser.no/, so I might be able to suggest improvements you could try.

If you prefer you can email me on post@amsleser.no. I would need to see electrical schematic and layout to give meaningful feedback.

dbeinder, 2023-04-27:
I have published my design here: https://github.com/dbeinder/han-mosquito

Last time I measured, I saw <50mV supply ripple when transmitting - not fantastic, but within the recommended limit according to the [HW design guidelines](https://www.espressif.com/sites/default/files/documentation/esp32-c3_hardware_design_guidelines_en.pdf).
I'm even getting this issue when powered from USB, so I think it shouldn't be an issue with the overall power budget. After factory reset, AP mode is very stable - when powered from USB, I can saturate the WiFi link with ~1MiB/s ping flood traffic and I see zero packet loss.

I'll have to hook up a scope close to the module power pins and see if there is a dip during the AP/STA switch. Maybe I can spot a difference between the affected units and the good ones.

ArnieO, 2023-04-27:
Very impressive size! 👍 
It is not so easy to measure the voltage ripples accurately, the current spikes are very short and strong.  And the buck converter could be injecting additional noise (more or less depending on your design and detail layout) into the ESP. 

Sorry to ask, but I did not find it: Is there a PDF of the electrical schematic somewhere on your Github?

dbeinder, 2023-04-27:
Thanks, creating compact layouts is my passion ;)

It should be possible to measure it with a short probe on the back directly at the vias.
You can open the project with KiCad 7, I didn't want to add PDFs to the repository.
Here's the schematic: [mosquito-c3.pdf](https://github.com/UtilitechAS/amsreader-firmware/files/11343296/mosquito-c3.pdf)

ArnieO, 2023-04-27:
Thank you! I don't have time right now to study your layout, or buck configuration - but based only on the schematic you could look into this as a first attempt:

- Add more capacitance on 3,3V side. I use 3 x 22µF on Pow-P1, and that is after a linear regulator (less noise than the switcher). Solder on top of C4 and C5. On P1 I have also a lot of low-ESR capacitance (ceramic) in front of the regulator (5V). The more the better.
- It is a general recommendation to always have a small (100 nF) cap were you use such large values as 22 µF. Smaller caps filter high frequencies better than large ones. Remember, a real world cap is a combination of C, R and L. The R and L parts are not unimportant for power filtering. Smallest cap closest to ESP.
- 1% might not be sufficient for R1/R2. Measure 3,3V level to see if there is correlation between voltage level and devices that do not make a stable AP. 
- Vcc/GND terminals on the ESP are in this area, C4/C5 should be night next to them. Do you have them on the back side, maybe?

Not related to the unstable AP issue:
- Probably a known tradeoff from your side, but I see you have omitted TVS diodes on the USB contact. So this design will not pass an EMC test.
- I would not use 35V rated C's at 36V, I always prefer more margin.

Noschvie, 2023-04-28:
@dbeinder do you plan a redesign?

gskjold, 2023-05-04:
Adding delay which in any case seems to make the firmware more reliable. 10ms extra wait when switching between AP ans STA is nothing.

ArnieO, 2023-05-04:
I understand your approach, and the tiny size of your device is impressive!

Bear in mind that the current spikes are very short and sharp, you may not be able to detect voltage fluctuations properly. 
Below are measurements done on a Seeeduino Xiao ESP32S3 running our code, using a [Nordic Power Profiler II](https://www.nordicsemi.com/Products/Development-hardware/Power-Profiler-Kit-2). The time span of the plot is 10 seconds, as you can see, the current spikes are short and sharp (high dV/dt), so any inductance between your capacitors and the chip will count. So unless you uncap the module and measure directly on the pin of the chip (which is maybe what you did?) you will not get the true picture. 

The datasheet seems to recommend external 10 µF + 100 nF, which is what I do on my designs, with those caps very close to the power terminals of the module.
[image]

[image]

[image]

[image]

[image]

closed 2023-06-11 by gskjold