UtilitechAS/amsreader-firmware #547: Show fractional digits on month plot

kimsey0, 2023-04-28:
The _Energy use last month (kWh)_ currently rounds the labels to show no fractional digits. For low-usage households, this means the numbers are pretty imprecise: Does "3" mean 2.5 or 3.49? There's a 40% difference between those two numbers.

[image]

I assume the reason is the limit on space for the label, but perhaps it could show a fractional digit only for numbers below 10? That is, maybe 

https://github.com/UtilitechAS/amsreader-firmware/blob/8b97544a2c7d4ecbaf3d4fa7f38b9090afe68250/lib/SvelteUi/app/src/lib/MonthPlot.svelte#L30

could change to `label: imp.toFixed(imp < 10 ? 1 : 0)`. Alternatively, it could use the new [Intl.NumberFormat](https://developer.mozilla.org/en-US/docs/Web/JavaScript/Reference/Global_Objects/Intl/NumberFormat) API with `maximumSignificantDigits` set to 2 or 3.

There's also the option of adding a more precise value as a mouse-over text, maybe even in addition to the change above?

ArnieO, 2023-04-28:
Thank you for the proposition - I think it is a good idea to have a minimum of two digits.
And if mouseover (with full resolution) is feasible (@gskjold ?), that would also be a nice improvement.

gskjold, 2023-04-28:
Agree! I've just never seen single digits on days :smile:

closed 2023-05-02 by gskjold

kimsey0, 2023-05-02:
Thanks a lot, @gskjold!