UtilitechAS/amsreader-firmware #552: Last day of april and first day of may have wrong data in graph

gskjold, 2023-05-02:
There seems to be an error where the month graph have wrong numbers for 30.april and 1.may. For me personally, it seems that a lot of the consumption from 30.apr have been included in 1.may.

Please report any similar findings

gskjold, 2023-05-02:
I'm realizing that it does not update values at all, and this is in fact values for 30.mar and 1.apr! Emergency release is coming soon!

gskjold, 2023-05-02:
Keeping this open a while for new visitors who have discovered this issue. This is fixed in v2.2.16

ArnieO, 2023-05-03:
**Note:**
The bug will lead to identical measurements being shown in the monthplot until v2.2.16 is installed, and the device has passed **_two_** midnights.

SirJMD, 2023-05-03:
> I'm realizing that it does not update values at all, and this is in fact values for 30.mar and 1.apr! Emergency release is coming soon!

Impressively fast response! :) 2.2.16 fixed it! Well done.

closed 2023-05-20 by gskjold